brainflow-dev/brainflow #18: How to deal with "error:the following arguments are required: --port, --board"

Hfut-Lujie, 2019-10-09:
[image]

I'm new to Python with argparse, can someone help me with where to add arguments?

Andrey1994, 2019-10-09:
you need to provide them, its a standard argparse module, for example for cyton on windows you need to run it as
python brainflow_get_data.py --port COM3 --board 0

Andrey1994, 2019-10-09:
just run it from command line outside Pycharm\Vscode... whatever you use

Andrey1994, 2019-10-09:
Also if you have specific board you can just remove argparse and hardcode values directly in python script, also there are SDKs for other languages so you can choose language you like

closed 2019-10-09 by Andrey1994

Hfut-Lujie, 2019-10-09:
Thank you very much. I'll have a try