brainflow-dev/brainflow #23: Start straming and then Segmentation fault error

redbaloon, 2019-12-22:
Dear Andrey,
I just found BrainFlow on GitHub and I think it is fantastic.
I run into problems: it crashes when it starts streaming.

I am a python user, Ubuntu 18.04, Python 3.8, Ganglion board with USB dongle BLED112.

`python main.py  --board-id 1 --serial-port /dev/ttyACM0  --mac-address e0:06:15:36:28:44 --log`
`[2019-12-22 19:29:33.146] [brainflow_logger] [info] incomming json: {`
`    "ip_address": "",`
`    "ip_port": 0,`
`    "ip_protocol": 0,`
`    "mac_address": "e0:06:15:36:28:44",`
`    "other_info": "",`
`    "serial_port": "/dev/ttyACM0"`
`}`
`[2019-12-22 19:29:33.146] [brainflow_logger] [debug] ganglionlib initialized`
`[2019-12-22 19:29:33.146] [brainflow_logger] [info] search for e0:06:15:36:28:44`
`[2019-12-22 19:29:35.117] [brainflow_logger] [debug] use stub streamer`
`[2019-12-22 19:29:35.340] [brainflow_logger] [debug] start streaming`
`Segmentation fault (core dumped)`

From: https://stackoverflow.com/questions/13654449/error-segmentation-fault-core-dumped

> "Segmentation fault (core dumped)" is the string that Linux prints when a program exits with a SIGSEGV signal and you have core creation enabled. This means some program has crashed.
As I am getting this error from running Python, this means the Python interpreter has crashed. There are only a few reasons this can happen:
> - You're using a third-party extension module written in C, and that extension module has crashed. 
> - You're (directly or indirectly) using the built-in module ctypes, and calling external code that crashes.
> - There's something wrong with your Python installation.
> - You've discovered a bug in Python that you should report.

I suppose we are in the first case.
Thanks in advice for your effort and time.

I wish you lovely holidays and a happy new year.
Kind Regards
Matteo Mazzanti

Andrey1994, 2019-12-22:
Hi,
I will take a look soon, I've not faced this issue before.
I dont have Ganglion with me right now and its the only board for which I didnt implement an emulator, so probably I will need to ask somebody to run C++ sample with GDB

redbaloon, 2019-12-22:
Hi.
Thanks for the prompt response.

Andrey1994, 2019-12-22:
in the meantime you can try prev version of brainflow for example 2.0.7(run pip install brainflow==2.0.7)

Andrey1994, 2019-12-22:
in 2.0.7 you should use 
```
board.start_stream ()
```
instead
```
board.start_stream (45000, args.streamer_params)
```

redbaloon, 2019-12-22:
> in 2.0.7 you should use
> 
> ```
> board.start_stream ()
> ```
> 
> instead
> 
> ```
> board.start_stream (45000, args.streamer_params)
> ```

Thanks, on it...

redbaloon, 2019-12-22:
Unfortunately, I got the same error:

`$ python main.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44 --log`
`[2019-12-22 20:38:49.110] [brainflow_logger] [info] incomming json: {`
`    "ip_address": "",`
`    "ip_port": 0,`
`    "ip_protocol": 0,`
`    "mac_address": "e0:06:15:36:28:44",`
`    "other_info": "",`
`   "serial_port": "/dev/ttyACM0"`
`}`
`[2019-12-22 20:38:49.110] [brainflow_logger] [debug] ganglionlib initialized`
`[2019-12-22 20:38:49.110] [brainflow_logger] [info] search for e0:06:15:36:28:44`
`[2019-12-22 20:38:51.369] [brainflow_logger] [debug] start streaming`
`Segmentation fault (core dumped)`
`$ pip freeze | grep brainflow`
`brainflow==2.0.7`

Andrey1994, 2019-12-22:
is seg fault in start_stream method or in another one? You can add prints between them to figure out, and sorry I have no board right now so I can not reproduce and debug it on my side

redbaloon, 2019-12-22:
After start_stream method:

[2019-12-22 22:12:43.159] [brainflow_logger] [debug] ganglionlib initialized
[2019-12-22 22:12:43.159] [brainflow_logger] [info] search for e2:31:cf:da:84:20
before board.start_stream 
[2019-12-22 22:12:45.407] [brainflow_logger] [debug] start streaming
after board.start_stream 
Segmentation fault (core dumped)

redbaloon, 2019-12-22:
PS: no rush. ;)

Andrey1994, 2019-12-28:
We will have an access to ganglion in the beginning of January and fix it then

Andrey1994, 2020-01-10:
Can not repr this issue on fedora/windows/macos. Will try to find ubuntu.
Could you try to run synthetic board to ensure that its ganglion specific?
Also it may help if you run C++ sample with gdb.
And to double check, your python and ubuntu both x64, right?

Andrey1994, 2020-01-10:
and one more question, are you sure that mac address is correct? For me it starts with e6:73:73 and I thought that this prefix is vendor specific and must be the same everywhere

Andrey1994, 2020-01-11:
Also can you run it wo mac address specified?

redbaloon, 2020-01-15:
Hi,
> Could you try to run synthetic board to ensure that its ganglion specific?
I will do this Saturday.

> Also it may help if you run C++ sample with gdb.
I don't know how to do this.

> And to double check, your python and ubuntu both x64, right?
yes

> Also can you run it wo mac address specified?
I copied the main.py example and I run it with a command that passes the mac address:
python main.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44
did I answered your question?

I will report on the first point asap.
Thanks

Andrey1994, 2020-01-15:
Try to run wo  --mac-address e0:06:15:36:28:44

redbaloon, 2020-01-20:
I run the synthetic board and all fine:
`
Hi,
I my computer crashed I had to reinstall the OS :)
Fresh installation Ubuntu 18.04

I did run the synthetic board
$ python main.py --board-id -1  --log
[2020-01-20 21:34:34.125] [brainflow_logger] [info] incomming json: {
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "other_info": "",
    "serial_port": ""
}
[2020-01-20 21:34:34.125] [brainflow_logger] [trace] prepare session
befofre board.start_stream 
[2020-01-20 21:34:34.125] [brainflow_logger] [trace] start stream
[2020-01-20 21:34:34.125] [brainflow_logger] [debug] use stub streamer
after board.start_stream 
[2020-01-20 21:34:34.125] [brainflow_logger] [info] noise range is -375:375
[2020-01-20 21:34:34.125] [brainflow_logger] [info] amplitude is 1000
[2020-01-20 21:34:34.125] [brainflow_logger] [info] shift is 0.3
[2020-01-20 21:34:44.140] [brainflow_logger] [trace] stop stream
[2020-01-20 21:34:44.143] [brainflow_logger] [trace] release session
[2020-01-20 21:34:44.143] [brainflow_logger] [trace] stop stream
[[ 0.00000000e+00  1.00000000e+00  2.00000000e+00 ...  1.96000000e+02
   1.97000000e+02  1.98000000e+02]
 [ 6.58024536e+02  7.53675232e+01  6.62902710e+02 ...  2.72592255e+02
   3.23583374e+02  4.25225525e+02]
 [ 2.25170044e+02  1.88392334e+01  5.74195068e+02 ... -1.14848022e+01
  -3.06759644e+00  4.84318695e+01]
 ...
 [ 3.56377075e+02 -1.66966171e+02  2.21182678e+02 ... -2.38129456e+02
  -1.61671997e+02 -2.71654938e+02]
 [ 3.09948181e+02 -2.58436127e+02 -3.24265350e+02 ... -2.83250000e+02
   8.35606384e+01  1.19407776e+02]
 [ 1.57955247e+09  1.57955247e+09  1.57955247e+09 ...  1.57955248e+09
   1.57955248e+09  1.57955248e+09]]
`
I don't know the command "wo --mac-address" I don't have it.

/Matteo

redbaloon, 2020-01-20:
`python main.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44 --log
[2020-01-20 21:47:29.530] [brainflow_logger] [info] incomming json: {
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "e0:06:15:36:28:44",
    "other_info": "",
    "serial_port": "/dev/ttyACM0"
}
[2020-01-20 21:47:29.530] [brainflow_logger] [debug] ganglionlib initialized
[2020-01-20 21:47:29.530] [brainflow_logger] [info] search for e0:06:15:36:28:44
befofre board.start_stream 
[2020-01-20 21:47:31.538] [brainflow_logger] [debug] use stub streamer
[2020-01-20 21:47:31.760] [brainflow_logger] [debug] start streaming
after board.start_stream 
Segmentation fault (core dumped)`

Andrey1994, 2020-01-20:
"wo --mac-address" means run it for ganglion using this command line
python main.py --board-id 1 --serial-port /dev/ttyACM0

Andrey1994, 2020-01-20:
I still can not reproduce it, if command from the comment above will also fail with segfault lets run c++ example with gdb, its the last idea I have

Steps:

```
sudo pip3 install cmake==3.13.0
git clone https://github.com/Andrey1994/brainflow
cd brainflow
https://github.com/Andrey1994/brainflow/blob/master/tools/build_linux.sh#L9 change Release here to Debug
bash tools/build_linux.sh
cd tests/cpp/get_data_demo
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=WRITE_FULL_PATH_TO_BRAINFLOW_FOLDER/installed_linux ..
# e.g for command above cmake -DCMAKE_PREFIX_PATH=/home/users/andrey/brainflow/installed_linux ..
make
sudo apt-get install gdb valgrind
gdb --args ./brainflow_get_data --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44
# gdb terminal will be opened type 'r' there
# after that there should be an error like in your python script
# type 'bt' in gdb terminal as soon as error occured and send output of 'bt' command 
```

redbaloon, 2020-01-20:
`python main.py --board-id 1 --serial-port /dev/ttyACM0  --log
[2020-01-20 23:20:11.652] [brainflow_logger] [info] incomming json: {
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "other_info": "",
    "serial_port": "/dev/ttyACM0"
}
[2020-01-20 23:20:11.652] [brainflow_logger] [debug] ganglionlib initialized
[2020-01-20 23:20:11.652] [brainflow_logger] [info] mac address is not specified, try to find ganglion without it
befofre board.start_stream 
[2020-01-20 23:20:13.724] [brainflow_logger] [debug] use stub streamer
[2020-01-20 23:20:13.946] [brainflow_logger] [debug] start streaming
after board.start_stream 
Segmentation fault (core dumped)`

redbaloon, 2020-01-24:
> I still can not reproduce it, if command from the comment above will also fail with segfault lets run c++ example with gdb, its the last idea I have
> 
> Steps:
> 
> ```
> sudo pip3 install cmake==3.13.0
> git clone https://github.com/Andrey1994/brainflow
> cd brainflow
> https://github.com/Andrey1994/brainflow/blob/master/tools/build_linux.sh#L9 change Release here to Debug
> bash tools/build_linux.sh
> cd tests/cpp/get_data_demo
> mkdir build
> cd build
> cmake -DCMAKE_PREFIX_PATH=WRITE_FULL_PATH_TO_BRAINFLOW_FOLDER/installed_linux ..
> # e.g for command above cmake -DCMAKE_PREFIX_PATH=/home/andrey/brainflow/installed_linux ..
> make
> sudo apt-get install gdb valgrind
> gdb --args ./brainflow_get_data --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44
> # gdb terminal will be opened type 'r' there
> # after that there should be an error like in your python script
> # type 'bt' in gdb terminal as soon as error occured and send output of 'bt' command 
> ```

I have done it.
The result:
```
(gdb) r
Starting program: /home/mrx/REPO/OBCI_BrainFlow/brainflow/tests/cpp/get_data_demo/build/brainflow_get_data --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2020-01-24 13:51:16.302] [brainflow_logger] [info] incomming json: {"ip_address":"","ip_port":0,"ip_protocol":0,"mac_address":"e0:06:15:36:28:44","other_info":"","serial_port":"/dev/ttyACM0"}
[2020-01-24 13:51:16.302] [brainflow_logger] [debug] ganglionlib initialized
[2020-01-24 13:51:16.302] [brainflow_logger] [info] search for e0:06:15:36:28:44
[2020-01-24 13:51:18.310] [brainflow_logger] [debug] use stub streamer
[New Thread 0x7ffff4b08700 (LWP 20009)]
[New Thread 0x7ffff4307700 (LWP 20010)]
[2020-01-24 13:51:18.531] [brainflow_logger] [debug] start streaming
[2020-01-24 13:51:18.531] [brainflow_logger] [info] Start sleeping in the main thread

Thread 2 "brainflow_get_d" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4b08700 (LWP 20009)]
0x00007ffff7bb2df4 in GanglionLib::read_message(int) () from /home/mrx/REPO/OBCI_BrainFlow/brainflow/installed_linux/lib/libBoardController.so
```

Andrey1994, 2020-01-24:
Thanks! We also managed to repr it on Ubuntu and debugging right now

redbaloon, 2020-01-24:
That is wonderful. Thanks!

redbaloon, 2020-01-28:
Wow! this is a great news!
Thanks thanks thanks!
/Matteo

Andrey1994, 2020-01-28:
does it work?

redbaloon, 2020-01-28:
:D 
I will check asap (Saturday probably) and let you know immediately.
/M.