brainflow-dev/brainflow #24: unable to stop streaming session

redbaloon, 2020-02-08:
Dear Andrey,
I am having a problem:
`
brainflow.board_shim.BrainFlowError: GENERAL_ERROR:17 unable to stop streaming session.
`
Using the python script provided at:
https://brainflow.readthedocs.io/en/stable/Examples.html
I am using:
`board.start_stream ()`
instead of:
`board.start_stream (45000, args.streamer_params)`

`$ python main.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address d2:e4:be:82:36:be --log   `
`[2020-02-08 17:01:46.795] [brainflow_logger] [info] incomming json: {   `
 `   "ip_address": "",   `
 `   "ip_port": 0,   `
 `   "ip_protocol": 0,   `
 `   "mac_address": "d2:e4:be:82:36:be",   `
 `   "other_info": "",   `
 `   "serial_port": "/dev/ttyACM0"   `
`}`
`[2020-02-08 17:01:46.795] [brainflow_logger] [debug] ganglionlib initialized   `
`[2020-02-08 17:01:46.795] [brainflow_logger] [info] search for d2:e4:be:82:36:be   `
`[2020-02-08 17:01:48.865] [brainflow_logger] [debug] use stub streamer   `
`[2020-02-08 17:01:49.087] [brainflow_logger] [debug] start streaming   `
`[2020-02-08 17:01:50.497] [brainflow_logger] [error] failed to stop streaming 13   `
`Traceback (most recent call last):   `
`  File "main.py", line 56, in <module>    `
`    main ()   `
`  File "main.py", line 49, in main    `
`    board.stop_stream ()    `
`  File "/home/mrx/anaconda3/envs/OBCI_Brainflow_213/lib/python3.8/site-packages/brainflow/board_shim.py", line 741, in stop_stream  `  
`    raise BrainFlowError ('unable to stop streaming session', res)`
`brainflow.board_shim.BrainFlowError: GENERAL_ERROR:17 unable to stop streaming session`

Thanks in advice for your help.
/Matteo M.

redbaloon, 2020-02-08:
The main.py script:
https://gitlab.com/mindlab.space/obci_brainflow-2.1.3_sandbox/-/blob/master/main.py

Andrey1994, 2020-02-08:
I can not repr it on fedora will try ubuntu

Andrey1994, 2020-02-08:
I will try to fix but if you need it to be working right now you can try another linux distribution it windows/macos

redbaloon, 2020-02-10:
Thanks. 
I am not in a hurry.
/Matteo

Andrey1994, 2020-02-14:
Could you try with 2.1.4? At least there should be another error code in logs and there is a chance that its resolved

matteo-mazzanti-ps, 2020-02-14:
`(OBCI_Brainflow) mrx@akira:~/REPO/OBCI_BrainFlow$ python main.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address e0:06:15:36:28:44 --log`
`[2020-02-14 16:35:56.221] [brainflow_logger] [info] incomming json: {`
`    "ip_address": "",`
`    "ip_port": 0,`
`    "ip_protocol": 0,`
`    "mac_address": "e0:06:15:36:28:44",`
`    "other_info": "",`
`    "serial_port": "/dev/ttyACM0"`
`}`
`[2020-02-14 16:35:56.221] [brainflow_logger] [debug] ganglionlib initialized`
`[2020-02-14 16:35:56.221] [brainflow_logger] [info] search for e0:06:15:36:28:44`
`[2020-02-14 16:35:58.313] [brainflow_logger] [debug] use stub streamer`
`[2020-02-14 16:35:58.536] [brainflow_logger] [debug] start streaming`
`[2020-02-14 16:36:14.590] [brainflow_logger] [error] failed to stop streaming 15`
`Traceback (most recent call last):`
`  File "main.py", line 56, in <module>`
`    main ()`
`  File "main.py", line 49, in main`
`    board.stop_stream ()`
`  File "/home/mrx/anaconda3/envs/OBCI_Brainflow/lib/python3.8/site-packages/brainflow`/board_shim.py", line 741, in stop_stream`
`    raise BrainFlowError ('unable to stop streaming session', res)`
`brainflow.board_shim.BrainFlowError: GENERAL_ERROR:17 unable to stop streaming session`
`(OBCI_Brainflow) mrx@akira:~/REPO/OBCI_BrainFlow$ pip show BrainFLow`
`Name: brainflow`
`Version: 2.1.4`
`Summary: Library to get data from OpenBCI boards`
`Home-page: https://github.com/Andrey1994/brainflow`
`Author: Andrey Parfenov`
`Author-email: a1994ndrey@gmail.com`
`License: UNKNOWN`
`Location: /home/mrx/anaconda3/envs/OBCI_Brainflow/lib/python3.8/site-packages`
`Requires: numpy`
`Required-by: `
`(OBCI_Brainflow) mrx@akira:~/REPO/OBCI_BrainFlow$`

matteo-mazzanti-ps, 2020-02-22:
Hi,
I have tried on other two different machines with Ubuntu 18.04. Same result.
Looking forward "try to fix ganglion on ubuntu #121".
/Matteo

closed 2020-03-02 by Andrey1994

Andrey1994, 2020-03-03:
There is a chance that its fixed in 2.2.0, at least for me it works now. But this fix was a dirty hack and probably smth is still wring. Feel free to reopen issue if it didnt solve the problem

matteo-mazzanti-ps, 2020-03-04:
It worked!