brainflow-dev/brainflow #36: Kernel assembly for MATLAB

masteroforest, 2020-05-04:
Good day.
I'm trying to build a kernel for MATLAB. The instructions for windows say: "Build it as a cmake project manually or use batch files from tools directory". Tell me, please, what should I collect: a release or the entire downloaded repository (containing the CMakeLists.txt file). It’s just that I just need to start the MATLAB scripts now. Also visual studio 2019: which commands for building a project should I run manually?

Best regards

Andrey1994, 2020-05-04:
Run this:
```
git clone https://github.com/brainflow-dev/brainflow
cd brainflow
# only if you use vs2019 instead vs2017
# open tools\build_win64.cmd with text editor
# change line "cmake -G "Visual Studio 15 2017 Win64" -DCMAKE_SYSTEM_VERSION=8.1 -DCMAKE_INSTALL_PREFIX=..\\installed64 .."
# to "cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=8.1 -DCMAKE_INSTALL_PREFIX=..\\installed64 .. "
# end of vs2019 specific instructions
# run batch file
.\tools\build_win64.cmd
```
It will compile and install core module after that follow instructions for matlab

Andrey1994, 2020-05-04:
Also you need to have cmake installed, you can install it from https://cmake.org/ or via
```
pip install cmake
```

closed 2020-05-04 by Andrey1994

masteroforest, 2020-05-04:
Thanks, I will try