brainflow-dev/brainflow #38: Buffer becomes empty after a while, and stays empty

khashdroid, 2020-05-06:
Hi there!
I'm having a great experience trying out brainflow with OpenBCI boards. Great framework! 
So far, There's only one issue that I have come across. After trying out brainflow on a windows PC, I moved to using a raspberry pi 3. I did compile everything from the source & double checked everything I did with the brainflow documentation. my program is meant to work on a real-time system, so basically the streaming continues forever unless the user wants to stop it. if I use the 'get_board_data' routine, everything goes well for about a few seconds up to a minute, and then then I only see empty data, and the 'data_count' is indeed zero.  if the 'get_current_board_data' routine is used, after a while the same data is emitted over and over (probably due to no new data entering the buffer). 
I have tested this with OpnBCI cyton & ganglion with the wifishield. 
This seems to be the case with both the python version and the C++ version of brainflow.

Andrey1994, 2020-05-06:
Hi,

If you use get_board_data method its ok that buffers gets empty from time to time since this method removes data from the buffer and returns it to you.

It looks like streaming gets stopped for some reason. I've seen such behaviour in Cyton Serial once and the problem was low battery. Unfortunately OpenBCI boards dont send battery level, so there is no way to check it. Just try to change batteries and lets hope that it will work.

<<This seems to be the case with both the python version and the C++ version of brainflow.
Yes, it should be the same since all bindings execute exactly the same code

Also, for debugging it will help if you send output with enabled logger and try to reproduce it using Cyton Serial instead WIFI Shield

Andrey1994, 2020-05-06:
BTW I am curious which OS you use with raspberry pi,I've not tested it...
Is it 32bit or 64?

Andrey1994, 2020-05-08:
Is it solved?

khashdroid, 2020-05-08:
Thanks for the swift reply!
- indeed seems that the streaming fully stops after a while. I do not think it is a battery problem issue, as the board does not stop streaming when I run the same code on a windows PC.... so I think it has something to do with the software running on the Rpi.
- unfortunately I do not have a serial dongle, so at the moment I can only test with the wifi shield.
- The OS on the Rpi is 32-bit (Armv7l), a.k.a raspbian buster

Andrey1994, 2020-05-08:
Before this commit https://github.com/brainflow-dev/brainflow/commit/fb80a246eb755fe6be6f70d2b2071d36c7db64e8  python binding didnt work on 32bit linux. But it doesnt affect C++

A few ideas for debugging, print "res" here https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/openbci/cyton_wifi.cpp#L35 if it's -1 add smth like that

```
#include <errno.h>

if (res == -1)
        {
#ifdef _WIN32
            safe_logger (spdlog::level::err, "WSAGetLastError is {}", WSAGetLastError ());
#else
            safe_logger (spdlog::level::err, "errno {} message {}", errno, strerror (errno));
#endif
        }
```

Andrey1994, 2020-05-08:
And one more test which may help - run synthetic board on raspberry device. If synthetic board works than its board specific issue

khashdroid, 2020-05-09:
so, I updated the repository (i.e. did a git pull), and now the command "`./tools/build_linux.sh`" fails in the end. here's what I see in the terminal:
`/home/pi/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E copy_if_different /home/pi/brainflow_dev/brainflow/compiled/libBoardController.so /home/pi/brainflow_dev/brainflow/python-package/brainflow/lib/libBoardController.so
Error copying file (if different) from "/home/pi/brainflow_dev/brainflow/compiled/libBoardController.so" to "/home/pi/brainflow_dev/brainflow/python-package/brainflow/lib/libBoardController.so".
make[2]: *** [CMakeFiles/BoardController32.dir/build.make:449: ../compiled/libBoardController32.so] Error 1
make[2]: *** Deleting file '../compiled/libBoardController32.so'
make[2]: Leaving directory '/home/pi/brainflow_dev/brainflow/build_lin'
make[1]: *** [CMakeFiles/Makefile2:170: CMakeFiles/BoardController32.dir/all] Error 2
make[1]: Leaving directory '/home/pi/brainflow_dev/brainflow/build_lin'
make: *** [Makefile:133: all] Error 2
`

khashdroid, 2020-05-09:
> And one more test which may help - run synthetic board on raspberry device. If synthetic board works than its board specific issue

Good idea! I will try that and get back to you with the outcome.

Andrey1994, 2020-05-09:
> so, I updated the repository and now the command "`./tools/build_linux.sh`" fails in the end. here's what I see in the terminal:
> `/home/pi/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E copy_if_different /home/pi/brainflow_dev/brainflow/compiled/libBoardController.so /home/pi/brainflow_dev/brainflow/python-package/brainflow/lib/libBoardController.so Error copying file (if different) from "/home/pi/brainflow_dev/brainflow/compiled/libBoardController.so" to "/home/pi/brainflow_dev/brainflow/python-package/brainflow/lib/libBoardController.so". make[2]: *** [CMakeFiles/BoardController32.dir/build.make:449: ../compiled/libBoardController32.so] Error 1 make[2]: *** Deleting file '../compiled/libBoardController32.so' make[2]: Leaving directory '/home/pi/brainflow_dev/brainflow/build_lin' make[1]: *** [CMakeFiles/Makefile2:170: CMakeFiles/BoardController32.dir/all] Error 2 make[1]: Leaving directory '/home/pi/brainflow_dev/brainflow/build_lin' make: *** [Makefile:133: all] Error 2 `

it looks like artefacts from previous build which cmaked failed to delte due to some reason(maybe library was loaded to a process or maybe permissions) but I am not sure and in CI we dont test 32bit linux

I will get raspberry device next week and try to reproduce/fix

Andrey1994, 2020-05-11:
> so, I updated the repository and now the command "`./tools/build_linux.sh`" fails in the end. here's what I see in the terminal:
> `/home/pi/.local/lib/python3.7/site-packages/cmake/data/bin/cmake -E [...]

This issue will be fixed in #43 I forgot to change lib name in several variables(for now you can run using branch from this PR or wait for this to be merged)

Andrey1994, 2020-05-11:
But I can not reproduce your original issue on raspberry pi, using this code I dont see zeroes
```
board.start_stream (45000, args.streamer_params)
for i in range (100):
    time.sleep(2)
    print(len(board.get_board_data()[0]))
board.stop_stream()
```
I use ganglion + wifi and it prints ~3200 all the time which makes sense since sampling rate is 1600

Andrey1994, 2020-05-11:
I can try to run your code if you provide a simple script to reproduce your issue

Andrey1994, 2020-05-19:
closing since I can not not reproduce and no updates

closed 2020-05-19 by Andrey1994