brainflow-dev/brainflow #45: Unable to build core.

masteroforest, 2020-05-16:
Hello! I tried several options for building the kernel ... 
1) Using Visual Studio 2019: changed the build_win64.cmd ("cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_INSTALL_PREFIX=..\\installed64 .. ") and launched it. Then I tried to open CmakeLists.txt from Visual Studio and compile a project on it.
2) Installed the visual studio 2017: launched an unchanged build_win64.cmd (from tools). Then I tried to open CmakeLists.txt from Visual Studio and compile a project on it.
Of course, cmake is installed, in all cases it started GUI first to check the equipment. I tried to build a kernel with both connected and disconnected boards(Cyton&Daisy). Screenshots below. Tell me, please, in how to assemble the core after all?

[image]
[image]
[image]

Andrey1994, 2020-05-16:
You need to do exactly whats written in docs and what I wrote in previous issues. You even dont need to launch visual studio.

Just open command line on windows, go to brainflow's folder and run tools\build_win64.cmd
[image]

In the end you will see this:
[image]

Thats all, core module is compiled and installed.

If you need Matlab binding you will see that now there are folders inc and lib in brainflow/matlab-binding/brainflow. 
https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#matlab

closed 2020-05-16 by Andrey1994

masteroforest, 2020-05-17:
Thank! The kernel has compiled. True, the file (build_win64.cmd) with the edits you specified moved to the root of the brainflow. Otherwise, the file could not find the CmakeLists.txt.