brainflow-dev/brainflow #46: MATLAB get data from CYTON&DAISY

masteroforest, 2020-05-17:
Good afternoon!
An example was launched with getting raw data from a SYNTHETIC_BOARD. In accordance with the documentation indicated Board Id: CYTON_DAISY_BOARD and BrainFlowInputParams.serial_port: COM3. Got an error (screen below). Tell me, please, how to fix it (perhaps somewhere there are examples not with a SYNTHETIC_BOARD)?

[image]

Best regards,
Dmitriy

Andrey1994, 2020-05-17:
For such questions we have a slack workspace, feel free to join. Its prefered since its not an issue with BrainFlow.

Serial port should be passed as a string, there are no variables\keywords to represent serial port. Also there is only default constructor for BrainFlowInputParams and after that you need to change fields. So, you need to write
```
params = BrainFlowInputParams ();
params.serial_port = 'COM3';
```

We dont have autogenerated docs for Matlab, but its exactly the same as for any other language, use docs for python as a reference

closed 2020-05-17 by Andrey1994

masteroforest, 2020-05-19:
Thank. I'll try to join a slack workspace. If successful, I will ask the following questions there.