brainflow-dev/brainflow #62: Connection problem with the openbci dongle

Heka18, 2020-06-26:
Hi, i just took an openbci helmet and i'm trying to use brainflow to connect with the dongle.
The problem is that when i try to connect i get the error: unable_to_open_port_error .
I'm on macOs 10.13.6 and I've already tried several things.
I tried to fix ftdi, the dongle is visible using the "ls -l / dev / tty *" command and also the openbci GUI connects without problems.

I think I miss something about the API but I can't understand what.
Any help is welcome

Andrey1994, 2020-06-26:
Could you provide a cmd line you are using? Also for macos serial port should be /dev/cu... (check docs)

Heka18, 2020-06-26:
With commands "ls /dev/cu.usb*" and "ls /dev/tty.usb*" I get the same code.
I've tried both with "dev/cu.usberial-xxxxxxxx" and with dev/tty.usbserial-xxxxxxxx , both don't work.

Moreover, the serial port is the same one to which the openbci GUI connects, and that work.

Andrey1994, 2020-06-26:
Share full command line you are using and output with enabled logger. Also try with sudo

Andrey1994, 2020-06-26:
And I hope GUI and other processes which use this port are closed. You can not open the same port twice, so close GUI and other apps if any, reconnect dongle and rerun

closed 2020-06-30 by Andrey1994