brainflow-dev/brainflow #77: integrating brainflow in Ue4

Freaknoizer, 2020-07-26:
Hi, first of all i wanted to thanks for your work, for sharing and maintain all this amazing project, its been really helpful!, i´m trying to set up unreal engine 4.25 to work with brainflow, i'm decided to make a custom build including brainflow library via Nuget package, (since I'm quite noob , i don't know if this is the best way to approach this or maybe build them as a plugin modules?), so i build the editor, but i went i wanted to call some function, it cant find the module,  i'm aware you already work with ue4 too, so i wanted to know a recommendation to go thru this, any light would be very helpful, im trying to use this using the emulator for testing a bci-cython board since right now i dont own one, (in there a way to integrate emulator in to unreal for testing?)

Andrey1994, 2020-07-26:
Hi!

Brainflow emulator is mostly for brainflow developers, it mimics data format from device and used mostly in CI to run tests automatically. And in fact its a test for brainflow. As a user you are supposed to use *BrainFlow Synthetic board* instead emulator. Here is a link: https://brainflow.readthedocs.io/en/stable/SupportedBoards.html#synthetic-board and since api is uniform you will be able to switch from synthetic board to real one easily. All examples for signal processing use synthetic board, feel free to check them: https://brainflow.readthedocs.io/en/stable/Examples.html#python-band-power

I am going to add a detailed guide for integration with unreal engine next week, roughly speaking you need to build brainflow c++ and link these libs to your game. Smth like here https://stackoverflow.com/questions/31039406/unreal-engine-4-linking-static-3rd-party-library-sdk-libzplay Its kinda build system written in C# inside unreal engine. After that you will be able to use brainflow in your C++ code for unreal engine game

There are a few tricks and details for this integration, I will document it soon

Andrey1994, 2020-07-26:
this may also help https://docs.unrealengine.com/en-US/Programming/BuildTools/UnrealBuildTool/ThirdPartyLibraries/index.html

Freaknoizer, 2020-07-26:
Thanks andrey for your answer, i will pick more in depth all the links, i'm really enthusiastic about this guide you are building! thanks a lot for your time and effort!

Andrey1994, 2020-07-27:
I've added instruction for integration with Unreal Engine. If you have further questions let's move to brainflow slack askhelp channel

Link to updated docs: https://brainflow.readthedocs.io/en/master/GameEngines.html#unreal-engine

closed 2020-07-27 by Andrey1994