brainflow-dev/brainflow #88: making changes in .java files

nosa999, 2020-09-20:
Hi,
I want to make some changes in .java files in this directory:
https://github.com/brainflow-dev/brainflow/tree/master/java-package/brainflow/src/main/java/brainflow
then, how can I compile them and make .class files? 
I use the command bellow in windows CMD, but its output has a lot of errors. I attached the picture of the errors in this question.

javac BoardShim.java

[image]
 this picture shows some part of the errors.

Andrey1994, 2020-09-20:
Hello, its maven project and you need to build it using maven commands like

```
mvn build
mvn package
....
```
More info here https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#java

And you need to install maven before that.
Also if you build it from source you need to build C++ code before java package from src. More info about C++ compilation here: https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#compilation-of-core-module-and-c-binding 

Maven getting started guide: https://maven.apache.org/guides/getting-started/

Andrey1994, 2020-09-20:
Also in java IDEs there are plugins for Maven you can use them instead command line

closed 2020-09-20 by Andrey1994