brainflow-dev/brainflow #90: Cyton Daisy WiFi only streams 8 channels

nateGeorge, 2020-09-27:
When streaming from the OpenBCI Cyton Daisy WiFi setup, I only get 8 channels of data (1-8).  Channels 9-16 are all 0s.  There was a similar issue in the OpenBCI GUI I reported (https://github.com/OpenBCI/OpenBCI_GUI/issues/837).  Here is the code I used to collect data:

```python
params = BrainFlowInputParams()
# default IP when connecting to board directly
params.ip_address = '192.168.4.1'
params.ip_port = 6227  # random port that was available
board = BoardShim(6, params)

board.prepare_session()
board.start_stream()

# other code is in here for collecting data
time.sleep(3)
data = board.get_board_data() # get all data and remove it from internal buffer
board.stop_stream()
board.release_session()

df = pd.DataFrame(data.T)
df.drop([0] + list(range(20, 30)), inplace=True, axis=1)
df.to_csv('test_data.csv', index=False)
```

If you examine the channels 1-8, they are fine, but channels 9-16 are all 0s.

I was using version 3.4.1 on Ubuntu.  I don't have the headset for a week or two so can't test the latest versions that were released, but it didn't look like anything was committed to the code that would fix this problem.

Andrey1994, 2020-09-27:
I am pretty sure its not an issue in code. Did you check your hardware setup?

nateGeorge, 2020-09-27:
Yes, before collecting data I tested that all channels were working in the GUI.  I know I've gotten wifi data working on all channels, but it was connecting directly to the board not using brainflow.

nateGeorge, 2020-09-27:
I will have to test again when I can, possibly next Wednesday, but if not, it may not be until a week or so later.

Andrey1994, 2020-09-27:
Which gui version did you use?

nateGeorge, 2020-09-27:
5.0.0 I believe.  It looks like actually the fixes were implemented in between 3.4.1 and the latest release, so I'll have to recollect the data with the latest version.  https://github.com/brainflow-dev/brainflow/commit/e564a2469f55939bea562434dca74a2de8332587

nateGeorge, 2020-09-27:
I think when I collected the data, 3.4.2 wasn't out yet and I didn't notice the data quality issue until just today.

closed 2020-09-27 by nateGeorge

Andrey1994, 2020-09-27:
GUI 5.0.0 uses brainflow and provides exactly the same data, so it should work

nateGeorge, 2020-09-27:
Yes, but I was using the brainflow API not the GUI to collect data.  Anyway, I noticed another issue with missing data from the API I'll post later probably (need to confirm it doesn't happen in the GUI).

Andrey1994, 2020-09-27:
GUI uses brainflow API and returned data is exactly the same everywhere.... in fact GUI runs exactly the same code as python scripts