brainflow-dev/brainflow #123: Add JS binding

Andrey1994, 2020-11-01:


GordonPmnt, 2023-07-10:
Hello guys,

I started working on it locally. Keep you posted

GordonPmnt, 2023-07-13:
Is there any kanban view or board somewhere? I'd like to put this in progress.

Already wrotte a pretty important part of the board-shim. My goal is to do it by the end of the year (cannot go very quick because I don't have much time), but definitely doing it.

Should I open a wip PR somewhere to show progress?

Andrey1994, 2023-07-13:
Wip pr will be good also I will be able to help then.

And we have no boards to track tasks

dman2210, 2023-07-18:
I might be able to pitch in here. Any progress?

GordonPmnt, 2023-07-18:
> I might be able to pitch in here. Any progress?

Actually yes @dman2210 . I did a good part of the board-shim and I started the data-filters.

I'm going to create a PR but for this I firstly need to find a good example to create a TS and JS package for brainflow. I'm not the king of webpack so I postponed a bit. If you have any available boilerplate, might help actually.

Here is a screenshot of a test I did with electron.js. Started a session with synthetic board, recorded 5 secs and did log the first channel. 
[image]

dman2210, 2023-07-18:
Is webpack necessary?

If so I am told these are good biolerplates to pull from for javascript webpack.
https://github.com/erikras/react-redux-universal-hot-example/blob/master/webpack/prod.config.js
https://github.com/pmmmwh/react-refresh-webpack-plugin/blob/main/examples/webpack-plugin-serve/webpack.config.js

Here are some package.json files as well for javascript.

https://github.com/erikras/react-redux-universal-hot-example/blob/master/package.json
https://github.com/pmmmwh/react-refresh-webpack-plugin/blob/main/package.json
https://github.com/firebase/firebase-js-sdk/blob/master/packages/app/package.json

This article had a very simple one for typescript.

https://blog.logrocket.com/using-webpack-typescript/

GordonPmnt, 2023-07-20:
Hello @dman2210 @Andrey1994 

This is here, I did put an example with the boardshim class: https://github.com/brainflow-dev/brainflow/pull/646/files

Right now, you can try it running the command `yarn run example` 
(you'll have to copy paste yourself the dll file in the lib repo though. I haven't updated any scripts yet.

Andrey1994, 2023-07-23:
Hi @GordonPmnt! Great progress, I will take a look

closed 2023-09-13 by Andrey1994