brainflow-dev/brainflow #135: DataFilter::read_file is broken on ARM32 (Raspberry Pi)

GrahamBriggs, 2020-11-15:
Build brainflow libs from latest source code on Raspberry Pi.

Try to run the C++ serialization sample program.

The execution enters DataFilter::read_file and never returns.

[image]

Andrey1994, 2020-11-15:
I can reproduce it, no idea why but I will try to add breakpoints inside read_file method and check

Andrey1994, 2020-11-15:
I think I've solved original issue but now I am getting memory error which I dont understand...
The issue is in EOF, need to change `c != EOF` by `!feof(fp)` here https://github.com/brainflow-dev/brainflow/blob/master/src/data_handler/data_handler.cpp#L794 and here https://github.com/brainflow-dev/brainflow/blob/master/src/data_handler/data_handler.cpp#L852

closed 2020-11-15 by Andrey1994