brainflow-dev/brainflow #148: Add support fort Muse devices

Andrey1994, 2020-11-29:
We can use BLED112 dongle and BGLIB to add support for Muse devices. Info about BLE services and characteristics can be taken from https://gitlab.com/brainbit-inc/brainbit-web

Use Ganglion Implementation as a referecence

xloem, 2021-01-24:
Does brainbit-web support the muse?
I've worked with the muse a little.  Here are the gatt characteristics: https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/constants.py
The serial characteristic accepts commands that initiate notification streaming on the others and configure which ones to stream.  most commands are used here: https://github.com/alexandrebarachant/muse-lsl/blob/master/muselsl/muse.py
The muse has hexadecimal 'profiles' that enable or disable different characteristics.  here are the profiles for the latest muse S; they are likely a superset of the older profiles: https://github.com/xloem/pymuse/blob/master/muse.py#L52

There are different kinds of muses:
- the original muse which was a BT serial device instead of BTLE (but accepted mostly the same commands)
- the new muses which are all BTLE: the muse 2016, muse 2, and muse S.  these work basically the same as each other but may have different different profile characteristics.

matthijscox, 2021-01-24:
We are looking for someone with a Muse and c/c++ experience to add it to BrainFlow. 

See the docs for some instructions https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html#instructions-to-add-new-boards-to-brainflow.

Andrey can also help you further if you are willing to volunteer ;)

Andrey1994, 2021-01-24:
"Does brainbit-web support the muse?"
Sorry, looks like I was writing two issues at the same time and copypasted smth wrong. It was about #147

Andrey1994, 2021-01-24:
I saw muse-lsl and going to use it as a reference, have no device to start working on it, maybe will get one soon

xloem, 2021-01-24:
I think the original muse-lsl research was borrowed from https://github.com/urish/muse-js .  There's also information at sdk archives like https://sites.google.com/a/interaxon.ca/muse-developer-site/home .  But the information in my first post is basically what's needed.

I might have some time to work on adding muse but I likely won't finish on my own (unless somebody can pay me or something ;p).  Is this repository a place to store partial work?

Andrey1994, 2021-01-24:
I can create a branch for muse and its ok if its in WIP state.

I wish I could pay for such work, but there is no company\research grant behind this project, so as of right now there is no such opportunity

xloem, 2021-01-24:
I'll try to find time to send some small PRs to a branch like that.  I'm a good bug generator, so it's good to have a separate branch.

Andrey1994, 2021-01-24:
I've created a branch for it https://github.com/brainflow-dev/brainflow/tree/muse

Note: this branch is created not from master. I used branch from this PR https://github.com/brainflow-dev/brainflow/pull/212

xloem, 2021-01-27:
@Andrey1994 I have an old muse where the contacts' electroplating had come off or something from extensive use.  Is that something that would be helpful to mail to somebody like you?

Andrey1994, 2021-01-27:
Yep it can help.
I think first priority should be adding support for Muse 2/Muse S, but original Muse device can be added as well.
Lets discuss in slack

Andrey1994, 2021-05-30:
Plan:

* add support for Muse S using BLED dongle(will work for all OSes) - https://github.com/brainflow-dev/brainflow/pull/287 
* refactor bglib part
* add support for Muse 2 using BLED dongle(will work for all OSes)
* add support for both boards using native BLE API(no dongle) for Windows 
* if there are requests from the users add support using native API for other OSes

dioptre, 2021-06-16:
Could you add LSL to BrainFlow and ignore the Muse specificity? Might be a good end to the other conversation of LSL vs Brainflow too ;)

Andrey1994, 2021-06-16:
It doesnt use LSL, it works directly with device without muse-lsl and it's good)

dioptre, 2021-06-16:
I know, but might be interesting to support both and show the difference ;)

Andrey1994, 2021-06-18:
PR merged to master branch, new release will be availabe soon. For further improvements I will create other tasks, closing this one

closed 2021-06-18 by Andrey1994