brainflow-dev/brainflow #151: Refactor Julia Code

matthijscox, 2020-12-02:
Good first issue for me; to refactor the Julia code with my current Julia knowledge.

I've started reading the Julia code, the first line I encounter is 
```
AnyIntType = Union{Int8, Int32, Int64, Int128, Int}
```
This is just weird... `Int` is the abstract supertype of the other types, you shouldn't have to declare a `Union`. Just try it in a Julia REPL:
```
julia> Int64 <: Int
true
```
So we can replace `get_timestamp_channel(board_id::AnyIntType)` simply with `get_timestamp_channel(board_id::Int)`

I can probably find more improvements, maybe even performance enhancements.

Andrey1994, 2020-12-02:
Nice, I didnt know it

Andrey1994, 2020-12-02:
Also, about refactoring, I dont like this https://github.com/brainflow-dev/brainflow/blob/master/julia-package/brainflow/src/board_shim.jl#L70 but if I move it to a function julia crashes

And enums in Julia as far as I understand work like defines in C++ and I dont like it at all and can not be duplicated in entire module. E.g https://github.com/brainflow-dev/brainflow/blob/master/julia-package/brainflow/src/data_filter.jl#L35 

Maybe there is better alternative to built-in enums

Andrey1994, 2020-12-02:
And there is a task to add Julia tests to CI system, its a part of https://github.com/brainflow-dev/brainflow/issues/146

matthijscox, 2020-12-02:
Yeah, I was looking at the `ccall` you were doing. I believe it's solved if you use a `const LIBRARY_PATH = "./lib/DataHandler.so"`
At least that's how we did it in another project.

Did you try using `const` ? If not, I will include this in my refactoring.

I also disliked enums, they are hardly used in Julia, nstead of enums everyone always just use subtypes:
```
abstract type DetrendOperations end
struct ConstantDetrend <: DetrendOperations end
struct LinearDetrend <: DetrendOperations end
```
This leads to maximum use of Multiple Dispatch, you can just dispatch on the desired type whenever you write a detrend function. I'll include it in the refactoring.

Andrey1994, 2020-12-02:
"Did you try using const ? If not, I will include this in my refactoring."
No, but if its a function which returns library name based on current OS can it be const?

My idea was smth like
```
lib_name = get_lib_name()
```

matthijscox, 2020-12-02:
yes, you can initialize the `const` that way. 

This was my first idea:
```
const DATA_HANDLER_INTERFACE = interface_path("DataHandler")
const BOARD_CONTROLLER_INTERFACE = interface_path("BoardController")
const ML_MODULE_INTERFACE = interface_path("MLModule")
```

Actually a `const` isn't of constant value, it's only of constant type.

matthijscox, 2020-12-02:
If we really get stuck, we should check how Julia does it automatically 

Browsing a bit, I see that in Julia 1.6 we'll even get faster Artifacts ;)
https://github.com/JuliaPackaging/JLLWrappers.jl/blob/master/src/toplevel_generators.jl#L39

An even better proposed way is to open the library once with [dlopen](https://docs.julialang.org/en/v1/stdlib/Libdl/#Libdl.dlopen) I believe. We can also provide the `dlopen` handle as constant interface I suppose, instead of a filepath.

Interestingly, the automated [init_library_product](https://github.com/JuliaPackaging/JLLWrappers.jl/blob/master/src/products/library_generators.jl#L54) does not use `const`, just a regular global handle.

I've forked the repo, and I'll make a PR, then you can test to see it still gives an error.

Andrey1994, 2020-12-02:
Not an expert in julia but I believe `ccall` doesnt open/close lib for each invocation.

This C++ library keeps the state and there are global board objects, I guess if you close the lib these objects will be removed(more likely). So in this case it would not work correctly, but I can be mistaken.

But I am agreed that dlopen is a good option and for other languages dlopen is used.

matthijscox, 2020-12-02:
Are the Julia tests in runtests.jl actually calling the c interfaces? I think I made a mistake, but I ran all tests and they give no errors or warnings.

Also they do not use the Julia test suite at all...

Andrey1994, 2020-12-02:
"But the tests do not use the Julia test suite at all..."
Its integration tests not unit tests, also exactly the same tests used as code samples in docs.

matthijscox, 2020-12-02:
I'll think about the tests, even integration tests can be done easily with the Julia test suite, which will be nice for summaries inside the REPL.

Anyway, I started a branch in my fork here: https://github.com/matthijscox/brainflow/tree/151_julia_refactor

Any Github administration you'd like me to do?

matthijscox, 2020-12-02:
I'll also start removing all those Integer() calls for the user:
`brainflow.enable_dev_brainflow_logger(brainflow.BOARD_CONTROLLER)`
instead of 
`brainflow.enable_dev_brainflow_logger(Integer(brainflow.BOARD_CONTROLLER))`

also we have a duplicate brainflow mention here, we could simplify to:
`brainflow.enable_dev_logger(brainflow.BOARD_CONTROLLER)`

Andrey1994, 2020-12-02:
I would like to keep it wo julia test suite...
These samples used here https://brainflow.readthedocs.io/en/stable/Examples.html#julia and its nice to keep these samples wo test related stuff and use them for testing to ensure that samples at docs working correctly.

Long time ago I used to duplicate tests and examples and it was in different files but it was much harder to keep up to date

<<Any Github administration you'd like me to do?
What do you mean?

matthijscox, 2020-12-03:
Are these pandoc examples automatically generated from the Julia test code? If so, pretty cool. (I'll not add any Julia Test usage then in these files)

It's been a while since I made a PR in Github. Is there something I should do extra? Or is there some workflow you prefer? E.g. I can make a [WIP] pull request already now, so you can add comments there.

Also, I the `const DATA_HANDLER_INTERFACE = interface_path("DataHandler")` works for me. I tried reproducing your issue and mentioned my solution in that [issue ](https://github.com/JuliaLang/julia/issues/29602) as well. But maybe you can also quickly try?

Andrey1994, 2020-12-03:
"Are these pandoc examples automatically generated from the Julia test code? "
Yes samples there are exactly these files with tests. 

In docs it looks like  
```
.. literalinclude:: ../julia-package/brainflow/test/brainflow_get_data.jl
   :language: julia
``` 

<<It's been a while since I made a PR in Github. Is there something I should do extra? Or is there some workflow you prefer? E.g. I can make a [WIP] pull request already now, so you can add comments there.

Nothing special, just create a PR and I will review it, lets keep it simple

<< But maybe you can also quickly try

I will try

matthijscox, 2020-12-03:
I'm getting creative with the library calls, dispatching on a type instead of an Integer(::Enum) pattern:
```
function set_log_level(log_level::Integer, log_lib::BrainFlowLib)
    ec = Integer(STATUS_OK)
    lib_cglobal = log_level_cglobal(log_lib)
    ec = ccall(lib_cglobal, Cint, (Cint,), Int32(log_level))
    if ec != Integer(STATUS_OK)
        throw(BrainFlowError(string("Error in set_log_level ", ec), ec))
    end
end

# still need to hardcode the cglobal input symbols, related to https://github.com/JuliaLang/julia/issues/29602
log_level_cglobal(::BoardControllerLib) = cglobal((:set_log_level, BOARD_CONTROLLER_INTERFACE)) 
log_level_cglobal(::DataHandlerLib) = cglobal((:set_log_level, DATA_HANDLER_INTERFACE)) 
log_level_cglobal(::MlModuleLib) = cglobal((:set_log_level, ML_MODULE_INTERFACE)) 
```

I think it's the best I can do so far, removing the need for all those Integer if/elseif checks.

I think we can also remove the error handling pattern with a 'decorator' macro, then we could write:
```
function set_log_level(log_level::Integer, log_lib::BrainFlowLib)
    lib_cglobal = log_level_cglobal(log_lib)
    @catch_brainflow_error("set_log_level") ccall(lib_cglobal, Cint, (Cint,), Int32(log_level))
end
```
That would clean up the code tremendously

matthijscox, 2020-12-09:
This issue can be closed once the [PR-154](https://github.com/brainflow-dev/brainflow/pull/154) is merged.

One thing for the future is to see if [Cbinding](https://github.com/analytech-solutions/CBinding.jl) and [CbindingGen](https://github.com/analytech-solutions/CBindingGen.jl) could further simplify the C API wrappers.

closed 2020-12-09 by Andrey1994