brainflow-dev/brainflow #166: Julia package registry

palday, 2020-12-15:
- [ ] Add JuliaRegistrator to GitHub integrations
- [ ] Make sure your `Project.toml` is up to date
    - [ ] is the version number correct? Should the bindings version number match the `brainflow` package number?
    - [ ] typically Julia package names begin with a capital letter (i.e. `Brainflow` instead of `brainflow`)
    - [ ] are you dependencies up to date?
- [ ] Comment on a commit  [`@JuliaRegistrator register subdir=julia-package/brainflow`](https://github.com/JuliaRegistries/Registrator.jl#registering-a-package-in-a-subdirectory) (or in an issue, if you add a `branch=master`)
- [ ] Profit!!!???

Also, for an example of packaging pre-built binary artifacts, see e.g. [MKL.jl](https://github.com/JuliaComputing/MKL.jl) or [LuaCall.jl](https://github.com/ararslan/LuaCall.jl) -- using artifacts or `_jll` packages is the way to go. (You can also use Conda to install things, as e.g. RCall does).

matthijscox, 2020-12-15:
We created an issue for [Create redistributable package for Julia - issue150](https://github.com/brainflow-dev/brainflow/issues/150). Would you be willing to help out?

I'm been trying to figure out how to create the artifact in the best way. Andrey has his own build pipeline that we need to connect to and he wants to keep everything in this single repo. Ill have a look at MKL.jl and LuaCall.jl.

Andrey is setting up a tarball. He currently only has a .zip file with the compiled libraries here: https://github.com/brainflow-dev/brainflow/releases/download/3.7.2/compiled_libs.zip

I'm getting some weird errors when I try to create an automatically downloadable artifact from that file. Now I'm considering to make a build.jl script or something that downloads it manually, and then creates the artifact locally from the downloaded files. That I could get that working in a script. But this doesnt feel like the preferred solution.

Andrey1994, 2020-12-15:
Yeah, thanks for instructions we will take a look. And its a duplication of #150, lets keep both open for now. 

I've already converted it to a tar in CI pipeline, but have yet to replace it in Release page. Will do it today

palday, 2020-12-18:
I can help out a bit, although it will probably be a few days until I get a chance to look at the entire build pipeline.

matthijscox, 2020-12-18:
Great! Please join the discussion in the other issue. We are now stuck with some artifact unpacking error on Windows

matthijscox, 2020-12-18:
I have begun a PR: https://github.com/brainflow-dev/brainflow/pull/172

Andrey1994, 2020-12-23:
https://github.com/JuliaRegistries/General/pull/26771

closed 2020-12-23 by Andrey1994

palday, 2020-12-23:
Cool.