brainflow-dev/brainflow #169: Create dockerfile with all packages preinstalled

Andrey1994, 2020-12-17:
Docker Image Specs:

* Use ubuntu 20 as a base image
* Install libgomp
* Make sure there is python3 and jdk >= 11
* Install julia and R
* Compile BrainFlow with -DUSE_OPENMP=ON(check #163 )
* Install all bindings from the local folder
* Also lets add some useful packages which are not required by BrainFlow(mne-python, pandas, etc)

For compilation add an option to provide git tag\commit\branch to checkout

I will update CI pipelines to build and publish image automatically later, for now just need to create a DockerFile

Stig124, 2020-12-18:
I'm actually working on it [https://github.com/Stig124/brainflow/tree/dockerfile](here)

Stig124, 2020-12-18:
I have managed to do something, I can't find a way to build R/Julia/Matlab binding procedurally, the other binding works great

Andrey1994, 2020-12-18:
Great!

Dont need to add matlab and I think its impossible at least because its not public software.

To prepare R environment you need to install libxml2-dev and r-base from package manager as I remember(maybe a few more), and before R package installation need to install python package. To install R dependencies you can use smth like

```
sudo -H R --vanilla -e 'install.packages("knitr", repos="http://cran.us.r-project.org")'
sudo -H R --vanilla -e 'install.packages("reticulate", repos="http://cran.us.r-project.org")'
```

For Julia you can take a look at this https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/run_unix.yml#L98 it access Julia terminal from bash

Feel free to create PR and I will take a look

closed 2020-12-19 by Andrey1994