brainflow-dev/brainflow #187: Add autodiscovery of serial ports for Cyton and Ganglion

Andrey1994, 2020-12-24:


ankur-gupta-29, 2023-07-25:
Hi I want to solve this problem .can you want little help from you.How can I proceed for this issue.and explain little more about this problem .

Andrey1994, 2023-07-27:
Sure and thanks!

There are some boards where we have to provide serial port via BrainFlowInputParams struct, e.g. cyton, freeeeg32, ganglion

If its not provided we return an error.

Instead error we can try to find serial port automatically, and it should be done for all 3 OSes

MaxMascini, 2024-11-13:
Any updates on this issue?

I've tinkered on it a bit and while searching through serial ports I keep getting an issue when trying to connect immediately after finding a compatible port where i often get:

` [2024-11-13 18:12:43.919] [board_logger] [error] board doesnt send welcome characters! Msg: BOARD_NOT_READY_ERROR:7 `

Sometimes it connects successfully, but the issue I'm having is for hyperscanning purposes where I'm trying to scan ports for and connect two boards within the same script. Occasionally both will successfully connect, but it is very inconsistent. 

Any ideas?

This is using OpenBCI Cyton Board(s), Windows 11, brainflow==5.14.0.

seanchangg, 2026-10-20:
Hi @Andrey1994  i think I'd like to take a shot at this? If it's still relevant

Andrey1994, 2026-10-20:
Hi, yes, it is, thanks!

closed 2026-11-02 by Andrey1994