brainflow-dev/brainflow #196: config_board in BoardShim class does not return confirmation message for OpenBCI commands on Cyton

florardu, 2020-12-29:
Hello,

I was trying to create a widget for the OpenBCI GUI, however, when calling the config_board command I do not receive any confirmation message in the form of a string. For example, sending the char "d" correctly resets the board, however, no message is sent back. This was done in Windows 10.

I then downloaded the Brainflow repo and wrote a small piece of code to verify that I could reproduce this problem using C++ code, and it still happened. This was done in Ubuntu.

Please find the code I wrote below:

```
int main() {

    BrainFlowInputParams params;
    params.serial_port = SERIAL_PORT;

    BoardShim::enable_dev_board_logger ();

    BoardShim* board = new BoardShim(CYTON_BOARD_ID, params);

    try {

        board->prepare_session();
        
        std::string conf = board->config_board("d");

        BoardShim::log_message ((int)LogLevels::LEVEL_INFO, conf.c_str());

    } catch (const BrainFlowException &err) {

        BoardShim::log_message( (int)LogLevels::LEVEL_ERROR, err.what() );
        if( board->is_prepared() ) {
            board->release_session();
        }
    }

    delete board;

}
```

Output:

```
[2020-12-29 11:19:51.515] [brainflow_logger] [info] incoming json: {"file":"","ip_address":"","ip_port":0,"ip_protocol":0,"mac_address":"","other_info":"","serial_number":"","serial_port":"/dev/ttyUSB0","timeout":0}
[2020-12-29 11:19:51.515] [brainflow_logger] [info] openning port /dev/ttyUSB0
[2020-12-29 11:19:53.641] [brainflow_logger] [warning] If you change gain you may need to rescale data, in data returned by BrainFlow we use gain 24 to convert int24 to uV
[2020-12-29 11:19:53.641] [brainflow_logger] [info] "THE CONFIRMATION MESSAGE SHOULD BE HERE."
```

I'm probably misunderstanding the way the config_board works, but I'm not sure.

Andrey1994, 2020-12-29:
There are no confirmation messages, if command sent there is no exception thrown. If command is not sent there will be an exception

closed 2020-12-29 by Andrey1994

Andrey1994, 2020-12-29:
For some boards there is a response in some cases for some of them there is not. For cyton its empty string. To check that command sent or not you should catch exceptions instead cheking return string