brainflow-dev/brainflow #197: unable to prepare streaming session

ironWolf1990, 2020-12-30:
Trying to stream data from Ganglion Board on Ubuntu 20.04 using the code [sample](https://brainflow.readthedocs.io/en/stable/Examples.html). Can someone help out how I can resolve this error.

```
python3 codesample.py --board-id 1 --serial-port /dev/ttyACM0 --mac-address d7:06:19:57:ee:88
[2020-12-30 16:07:32.319] [brainflow_logger] [info] incoming json: {
    "file": "",
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "d7:06:19:57:ee:88",
    "other_info": "",
    "serial_number": "",
    "serial_port": "/dev/ttyACM0",
    "timeout": 20
}
[2020-12-30 16:07:32.319] [brainflow_logger] [debug] use dyn lib: /home/wolf/.local/lib/python3.8/site-packages/brainflow/lib/libGanglionLib.so
[2020-12-30 16:07:32.319] [brainflow_logger] [trace] Board object created 1
[2020-12-30 16:07:32.319] [brainflow_logger] [debug] Library is loaded
[2020-12-30 16:07:32.319] [brainflow_logger] [info] use 20 as a timeout for device discovery and for callbacks
[2020-12-30 16:07:32.319] [brainflow_logger] [debug] ganglionlib initialized
[2020-12-30 16:07:32.319] [brainflow_logger] [info] search for d7:06:19:57:ee:88
[2020-12-30 16:07:32.319] [brainflow_logger] [error] failed to Open Ganglion Device 1
Traceback (most recent call last):
  File "codesample.py", line 52, in <module>
    main ()
  File "codesample.py", line 39, in main
    board.prepare_session ()
  File "/home/wolf/.local/lib/python3.8/site-packages/brainflow/board_shim.py", line 848, in prepare_session
    raise BrainFlowError ('unable to prepare streaming session', res)
brainflow.board_shim.BrainFlowError: GENERAL_ERROR:17 unable to prepare streaming session
```

Andrey1994, 2020-12-31:
Hi, it looks like permission issue with serial port. Run with sudo or configure permission for your user. Also, make sure that serial port is correct and try wo mac address first

ironWolf1990, 2020-12-31:
Hey @Andrey1994 , thanks for the reply. I added proper group [permissions](https://askubuntu.com/questions/58119/changing-permissions-on-serial-port). I also tried running it without mac address , still having the same issue. I have no problems as such when connecting thought the `OpenBCI_GUI`. 

Suggestions ?

Andrey1994, 2020-12-31:
Run with sudo, make sure that there are no other processes using this port(close gui, etc)

ironWolf1990, 2020-12-31:
@Andrey1994 , I was able to run my script using `su - wolf -c "python3 /home/wolf/codebase/python/openbci-finger-signal/codesample.py --board-id 1 --serial-port /dev/ttyACM0 --log"`.

Is there a fix for this? I might have issues if i create a `.ipynb` notebook !

Andrey1994, 2020-12-31:
You just need to configure permissions for it correctly. Its a common task for unix like systems and not part of this project

closed 2020-12-31 by Andrey1994

Andrey1994, 2020-12-31:
Or maybe you just need to logout and login again after adding your user to this group

ironWolf1990, 2020-12-31:
On Linux/Ubuntu if anyone comes across this issue , permissions for `/dev/ttyACM0` can be permanantly solved by adding yourself to the `dialout` group -
1. `sudo usermod -a -G dialout $USER`
2. `sudo adduser $USER dialout`

Restart your computer after executing the above commands. Just `Logout` doesn't do the job.

Andrey1994, 2021-01-01:
Good to know, I supposed logout should be enough(or maybe reconnect the dongle)