brainflow-dev/brainflow #207: Exception during board connection Ganglion

rmib200, 2021-01-09:
While running the Python code example for Getting Data from Board I get the following traceback Error.
Could anyone help with this issue please?
I'm copying the full traceback.

`---------------------------------------------------------------------------
SystemExit                                Traceback (most recent call last)
<ipython-input-15-4b2e3109ee4f> in <module>
     47 
     48 if __name__ == "__main__":
---> 49     main ()

<ipython-input-15-4b2e3109ee4f> in main()
     14     parser.add_argument ('--file', type = str, help  = 'file', required = False, default = '')
     15     parser.add_argument ('--log', action = 'store_true')
---> 16     args = parser.parse_args ()
     17 
     18     params = BrainFlowInputParams ()

d:\ProgramData\Anaconda3\envs\openbci\lib\argparse.py in parse_args(self, args, namespace)
   1753     # =====================================
   1754     def parse_args(self, args=None, namespace=None):
-> 1755         args, argv = self.parse_known_args(args, namespace)
   1756         if argv:
   1757             msg = _('unrecognized arguments: %s')

d:\ProgramData\Anaconda3\envs\openbci\lib\argparse.py in parse_known_args(self, args, namespace)
   1785         # parse the arguments and exit if there are any errors
   1786         try:
-> 1787             namespace, args = self._parse_known_args(args, namespace)
   1788             if hasattr(namespace, _UNRECOGNIZED_ARGS_ATTR):
   1789                 args.extend(getattr(namespace, _UNRECOGNIZED_ARGS_ATTR))

d:\ProgramData\Anaconda3\envs\openbci\lib\argparse.py in _parse_known_args(self, arg_strings, namespace)
   2020         if required_actions:
   2021             self.error(_('the following arguments are required: %s') %
-> 2022                        ', '.join(required_actions))
   2023 
   2024         # make sure all required groups had one option present

d:\ProgramData\Anaconda3\envs\openbci\lib\argparse.py in error(self, message)
   2506         self.print_usage(_sys.stderr)
   2507         args = {'prog': self.prog, 'message': message}
-> 2508         self.exit(2, _('%(prog)s: error: %(message)s\n') % args)

d:\ProgramData\Anaconda3\envs\openbci\lib\argparse.py in exit(self, status, message)
   2493         if message:
   2494             self._print_message(message, _sys.stderr)
-> 2495         _sys.exit(status)
   2496 
   2497     def error(self, message):

SystemExit: 2

usage: ipykernel_launcher.py [-h] [--timeout TIMEOUT] [--ip-port IP_PORT]
                             [--ip-protocol IP_PROTOCOL]
                             [--ip-address IP_ADDRESS]
                             [--serial-port SERIAL_PORT]
                             [--mac-address MAC_ADDRESS]
                             [--other-info OTHER_INFO]
                             [--streamer-params STREAMER_PARAMS]
                             [--serial-number SERIAL_NUMBER] --board-id
                             BOARD_ID [--file FILE] [--log]
ipykernel_launcher.py: error: the following arguments are required: --board-id
An exception has occurred, use %tb to see the full traceback.

SystemExit: 2
`

Andrey1994, 2021-01-10:
Hi. You didnt provide required argparse arguments(read about argparse in python docs) and argparse intended to launch from cmd not from juputer

closed 2021-01-10 by Andrey1994

rmib200, 2021-01-10:
What are those arguments? Is brainflow not intended to work on jupyter notebooks?

Andrey1994, 2021-01-10:
It works in jupyter, sample which you copypasted intended for cmd run, not for jupyter. Check docs for information about this arguments