brainflow-dev/brainflow #214: Build process produces files outside build folder

xloem, 2021-01-24:
I probably haven't found where this quirk is documented and explained, but usually with cmake projects it is conventional to keep all build products within the build folder, of course like this:
```
mkdir build
cd build
cmake ..
make
# built files are now all within ./build/
```
This lets multiple build folders be made, for e.g. debug vs release or other changes to the build configuration or sharing a source folder across machines.

In brainflow, this is not the case: the build products are produced in the source directory, somehow.

Andrey1994, 2021-01-24:
in BrainFlow some build artifacts(compiled dyn libs) copied from build folder to the bindings automatically, idea is that bindings need these libs and these libs should be in binding's search path when you build python whl or java jar, etc.

You still can build debug or release configurations and there is no issue with it but both libs will have the same name.

closed 2021-01-24 by Andrey1994

xloem, 2021-01-24:
Cool =)

I think with cmake usually you would set up an installation folder, use `make install`, and then reference the install folder from the bindings.  But it sounds like this isn't something you find valuable; it really makes little difference.

Andrey1994, 2021-01-24:
bindings are not aware of installation folder... and not for all of them its possible to pass this info

Andrey1994, 2021-01-24:
and I guess debug vs release makes sense mostly for C++ developers, and you can create two different installation folders using cmake, smth like:
```
-DCMAKE_INSTALL_PREFIX=$GITHUB_WORKSPACE/installed_debug
-DCMAKE_INSTALL_PREFIX=$GITHUB_WORKSPACE/installed_release
```

xloem, 2021-01-24:
I've worked with the source a little more and I see the design issue around what to put in cmake and what to put in the bindings config systems.  It looks like you've made choices that help the project move forward effectively and rapidly.