brainflow-dev/brainflow #227: Remove Julia transpose()

matthijscox, 2021-02-10:
Julia has column-major indexing and matrix memory layout, so data matrix transpose in `get_board_data` like in Python is not needed.

We decide to remove it. Breaking interface change.

This also impacts a few data filtering functions that take the data matrix as input.

I can also cleanup some other things that I found.

matthijscox, 2021-02-10:
Another possible breaking change for review by @Andrey1994: Julia uses the ! as style for mutating functions.

Some functions mutate the input data, like `detrend`, which will thus be renamed to `detrend!`. Other functions make an internal copy of the input, like `calc_avg_band_powers`, those will remain the same (though I will remove some double copying behavior I saw)

The user now immediately knows which functions are mutating and which are not.

However, it's not a style used in other language bindings, so I leave the decision to Andrey.

Andrey1994, 2021-02-10:
As I understand ! is a style guide, not a requirement.

Lets keep function names consistent with other languages, specially since we have no docs for julia

Andrey1994, 2021-02-10:
Btw is column major also a case for matlab? Should we change it there?

matthijscox, 2021-02-10:
Ah yes, MATLAB is also column major... I can try to change as well (though I have no license at home), but Julia has my priority ;)

matthijscox, 2021-02-11:
R is also column-major I see

Andrey1994, 2021-02-11:
For R we call python code, lets keep as is

closed 2021-03-04 by Andrey1994