brainflow-dev/brainflow #230: get_board_data should have an overload allowing to specify number of samples to read

glopesdev, 2021-02-11:
The C API allows reading and flushing from the ringbuffer with a specified number of samples and into a pre-allocated memory buffer. Both of these are important for efficiently sampling streaming data.

The fix would be to provide an overload for `get_board_data` which accepts an external buffer, i.e. change the signature here: 
https://github.com/brainflow-dev/brainflow/blob/5fc42b51b90611931219eaca76eb1667fa575887/csharp-package/brainflow/brainflow/board_shim.cs#L692

for example, with this signature (and example implementation):
```c#
        public void get_board_data (double[] data_arr)
        {
            int num_rows = BoardShim.get_num_rows (master_board_id);
            int size = data_arr.Length / num_rows;
            int ec = BoardControllerLibrary.get_board_data (size, data_arr, board_id, input_json);
		    if (ec != (int) CustomExitCodes.STATUS_OK) {
```

There are other possible, roughly equivalent signatures and implementations. Any solution would be fine as long as it is possible to provide the sample buffer for the method externally, to avoid allocations, and to allow fixing the size of data extracted from the ring buffer.

Andrey1994, 2021-02-11:
Hi,

to allow fixing the size of returned data I think API should look like:
```
get_board_data(int num_samples)
```
But then the questions I have:

- from which end of the queue it should return data?(FIFO of FILO)
- how should it work if there is less data than you requested? API is non blocking and for exampele for get_current_board_data it returns less datapoints 

Why do you need external buffer instead returned by the function?
Do you have any perf issues with current approach?

matthijscox, 2021-02-11:
I think he just wants to provide a pre-allocated array and overwrite it's elements.
Also I think maybe we're talking about `get_current_board_data` ?

Now `get_board_data` and `get_current_board_data` create their own pre-allocated array inside the bindings. 

In Julia this may also be a nice easy feature to add, I would just make something like:
```
function get_current_board_data(data::Matrix, board_shim::BoardShim)
    num_rows, num_samples = size(data)
    # rest of the code
end
```

Not that I have seen any performance issues, but it would be a small improvement.

glopesdev, 2021-02-11:
@matthijscox yes, that was the request exactly, to provide a pre-allocated array. There is no need to change or decide on LIFO or FIFO size, since this maps directly to the C/C++ API.

`get_current_board_data` provides this functionality, but doesn't flush the ringbuffer and only returns the latest elements.

On embedded devices this would allow creating allocation-free data processing pipelines, to minimize unnecessary GC.

Andrey1994, 2021-02-11:
"There is no need to change or decide on LIFO or FIFO strategy, since this maps directly to the C/C++ API."

No, there is such need.

Data is stored in a ring buffer internally which has 2 ends. And right now `get_current_board_data` returns data from the end(FILO). While `get_board_data` in fact FIFO but returns all data, so it doesnt matter right now. But it will be if we decide to return only a chunk.

It sounds like two potential requests:

- change `get_current_board_data` to push data to external buffer wo flush
- change `get_board_data` to return single chunk of data instead of entire array

About preallocated arrays: its a 2d array and users will need to prepare it properly...

glopesdev, 2021-02-11:
@Andrey1994 I'm not sure I follow. From looking at the implementation at `get_board_data`, the P/invoke function which is called from the C API has exactly the signature which I suggested.

The only thing the current implementation seems to do is to automatically compute the chunk size using `get_board_data_count`, but other than that it is already operating entirely on chunks, except the allocation of those chunks is done inside the function. I don't see how this would ever be a problem if I preallocated the arrays outside and checked `get_board_data_count` to make sure there is enough data to feel the buffer. Or am I missing something here?

In fact, for now I ended up simply calling `BoardControllerLibrary.get_board_data` in this way and it works fine. If the underlying C API is a true ringbuffer, then I don't understand how this would ever be an issue.

Also, @matthijscox mentioned `get_current_board_data` but I never mentioned this API other than to make a comparison, so to clarify, I am only interested in `get_board_data`, so we could keep this only to that issue.

Regarding the shape of the array, I would be fine even just using a flattened 1D array. These APIs are useful for low-level performant code, where developers are supposed to know what they are doing anyway, so I don't think this would be a problem.

glopesdev, 2021-02-11:
On second thought, assuming that `BoardControllerLibrary.get_board_data` will remain public, it might be enough for low-level API access purposes, I don't think there is the need for anything extra (other than maybe throwing an exception).

Andrey1994, 2021-02-11:
"the P/invoke function which is called from the C API has exactly the signature which I suggested."

In the low level API it's a 1d array because you can not pass NDArrays from other languages to C\C++, but low level API is not designed for the users, its for bindings.

There are wrappers on top of low level code, and users expected to work with BoardShim class, if you take a look at the signature of get_board_data in BoardShim class you will notice that it returns 2d array. So if you want to use preallocated array you need to preallocate 2d array.

And inside `get_board_data ()` I will still need to create 1d array temporary and reshape it to 2d array.

"On second thought, assuming that BoardControllerLibrary.get_board_data will remain public"

yep, it will be public as well as all methods of this class

glopesdev, 2021-02-12:
I understand the limitation around 1D array marshalling does lead to some awkwardness. I believe it would be possible to work around it by using the `fixed` keyword to take a pointer, but that might have its own complications.

I think `BoardControllerLibrary.get_board_data` is probably a good solution for lower-level API developers, so this issue can probably be closed for now.

closed 2021-02-12 by Andrey1994

Andrey1994, 2021-02-12:
btw are you trying to integrate brainflow into https://bonsai-rx.org/ ?
it looks great, let me know if you implement it or need any help

glopesdev, 2021-02-12:
yes, that's it :) there is already a first working draft at the below repository, which I've tested successfully on a Cyton board:
https://github.com/bonsai-rx/brainflow

Will ping in here if I hit any other limitations. So far we were trying to reduce the latency receiving data from Cyton as much as possible, but it looks like we can't go below ~250-500ms or so. Is there a gitter/slack chat where it might be better to discuss these things? thanks

Andrey1994, 2021-02-12:
there is a slack - https://c6ber255cc.execute-api.eu-west-1.amazonaws.com/Express/ 
for cyton latency take a look at this link - https://docs.openbci.com/docs/10Troubleshooting/FTDI_Fix_Windows its specific for cyton and can not be handled by brainflow

Andrey1994, 2021-02-12:
it would be great if you implement it for all supported boards not only for cyton