brainflow-dev/brainflow #234: add methods to remove noise at 50(EU) and 60(USA) hz

Andrey1994, 2021-02-14:
there is a bandstop filter, but it requires multiple arguments, we can add helpers to remove noise for EU and USA users wo knowledge about optimal band width, order and filter type

matthijscox, 2021-02-15:
Let's figure out the defaults here first

For bandstop I used:
    center_freq::Float64 = 50.0 or 60.0 (depending on EU or USA)
    band_width::Float64 = 4.0
    order::Integer = 4
    filter_type::Integer = BUTTERWORTH
    ripple::Float64 = 0.0

Andrey1994, 2021-02-15:
these values are fine as for me, I am also interested is it possible to determine timezone in C\C++ automatically for all OSes. Then maybe eu\usa can be optional arguments

Andrey1994, 2021-02-15:
smth like:

```
enum EnvNoise (enum.Enum):
    AUTO = 0
    FIFTY = 1
    SIXTY = 2
    
def remove_environmental_noise (env):
.......
```

closed 2021-04-11 by Andrey1994