brainflow-dev/brainflow #246: shitty misinformation about LSL published by BrainFlow

dmedine, 2021-02-26:
Hi. I am one of the maintainers of the LSL project and I was extremely disheartened to read one of your recent blogposts (https://brainflow.org/2021-02-05-lsl-review/) which makes numerous false and misleading claims about LSL. I am extremely puzzled by why you have chosen to do this. LSL and BrainFlow serve different purposes and could, in fact, compliment and enhance one another when used in combination. This blogpost, however, paints LSL as an inferior alternative to BrainFlow. Why you have chosen to dishonor our little tool is a mystery to me, but many of the claims you make about LSL range from misleading to downright incorrect. Furthermore, this blogpost does a great disservice not only to LSL, but also to BrainFlow! Since the post contains so many inaccurate statements about LSL, it makes LSL look bad to people who don't know anything about these topics, and it makes _you look bad_ to anyone who does!

I shall address these issues one by one.

> LSL works on top of TCP\IP stack, so it means that devices, which use BT, BLE or serial port for communication can not stream LSL data directly. 

This is not accurate. LSL _is an implementation of a TCP\IP stack_ with a specific protocol common to all LSL streams. This is all semantics, of course. In terms of process, it is important to understand that in order to stream information from a BT, BLE or serial enabled device, one must first communicate with the device driver and then forward the information decoded by the driver to whatever the next layer of software is (e.g. LSL or BrainFlow or whatever). Your software communicates with device drivers in exactly the same manner. The difference is simply that while LSL forwards the device information to a TCP\IP implementation, yours forwards it to a library which provides developers with a common set of wrapper functions to methods like 'get_board_data' etc. In that sense _devices which use BT, BLE, or serial port cannot stream data using BrainFlow directly_ either!

> Also, many boards use Arduino which is not supported by LSL.

This is simply wrong. Although there is not currently an out of the box Arduino solution for LSL on github, it is possible to use LSL from within an Arduino development environment to acquire and stream data. I have done it.

> Such pipeline works fine for researchers, but not ideal for software engineers, because it’s difficult to automate it, not perfect from UX point of view and requires sophisticated Inter Process Communication.

This statement  is misleading if not entirely false. While I wouldn't presume to assume what is or is not 'ideal for software engineers', I do know that dozens of commercial software projects as well as many industrial level scientific computing systems (e.g. groups within NASA) are currently using LSL. Thus I would argue that it is indeed quite useful (if not ideal) for many developers who don't have time to muck about with kludgey research-grade code-bases. In fact the people who work a lot on LSL work very hard to make this the case. Your saying anything to the contrary is not nice to them!

I don't quite understand what you find difficult to 'automate' about LSL. Many core LSL projects on our github page support CI/CD quite fully and quite smoothly. There are also numerous ways to 'remote control' various LSL apps. LSL is highly customizable with configuration files that can be used either globally or locally. Beyond that, I don't know what level of automation you expect from a software framework. 

You are right in observing that LSL is not geared towards naive end-users, but that is not the point of LSL (it is a backend after all); but, one of the stated design goals behind LSL _is_ to simplify inter-process communication. This it does quite well. For example, it requires no thread handling or memory management whatsoever to have one LSL app that listens to a device and transmits its data, another one that receives that data and does some kind of analysis, and a third that listens for the output of the analysis and uses it as a control parameter to some separate hardware or software device. In fact, because LSL handles the all the details of asynchronous I/O and memory management under the hood, each of these processes could function equally well on different computers or as separate parts of the same application. Thus the developer is free to focus on internal processes themselves rather than the problems associated with thread/memory management like inter-process communication, race-conditions, deadlock, indeterminacy, and so on. In that sense, using LSL actually takes a huge amount of programming overhead away from developers.

> Additionally, LSL doesn’t allow sending/receiving commands to/from a device, which means its state cannot be changed during streaming. This lack of communication with the device can also cause problems with timestamp synchronization, since LSL has no information about the transmission period from the device to the LSL application.

This is actually true but it is a rather misleading statement. LSL does not (usually) run on embedded hardware, so how could it possibly ever communicate directly with a device? Again, to talk to a device, as I stated above, one needs to go through the device driver. When acquiring information from a hardware device connected to a PC, there is always a digital bit stream protocol, no matter how generic, and that encoded information must be decoded by a driver. Your software relies on this inescapable fact as well! 

As for your claim that this somehow causes synchronization problems, you could not possibly be more false and insulting about LSL. No matter what software stack you use---be it LSL, BrainFlow, or anything---in order to properly synchronize data from different devices, one must first know the device latency. This must be measured and there are techniques for doing it (some notable [examples of which](https://www.neurobs.com/manager/content/docs/labstreamer/index.html#0) are actually powered by LSL!). Since your API doesn't have any functions for timestamping, synchronizing, or even handling multiple data streams simultaneously, you would be wise to keep silent on this topic.

Furthermore, I am quite familiar with some of the hardware you support and I know for a fact that when those devices are in acquisition mode, the device parameters cannot be changed. Given that fact, it is especially unfair to criticize LSL on this point.

> LSL was originally designed as a tool to synchronize data streams. It is not intended to be a replacement for device-specific SDKs. In contrast, BrainFlow is designed to be a device agnostic SDK for a growing list of supported devices, which standardizes the data stream at the software level

This is true. This is a fair and fine point. I completely agree with this statement.

In conclusion, I very much don't want to get into a prolonged dispute, but you may not defame our organization with articles like this. Many of us have put countless hours into honing this tool as a service to the science and engineering communities that we are a part of. These people don't get paid for their valuable efforts and they work very hard and very well. Thus for their sake if not your own, please remove this dreadful blogpost as soon as possible!

Andrey1994, 2021-02-26:
Hi,

```
Your software communicates with device drivers in exactly the same manner. 
```

No, its not true and it's the main difference. BrainFlow is a device driver itself. And works directly with device without any drivers\libs\apps. While LSL needs another app to communicate with a board.

```
In that sense devices which use BT, BLE, or serial port cannot stream data using BrainFlow directly either!
```

It's also wrong, as I mentioned above it communicates directly with the board and if device sends data via BLE\BT, implementation of this board in BrainFlow will read\send data directly from\to device. Also, there are examples, helpers and **existing** devices supported by BrainFlow which work over BLE.

```
Although there is not currently an out of the box Arduino solution for LSL on github, it is possible to use LSL from within an Arduino development environment to acquire and stream data. I have done it.
```

It's great that you did it, but firmware engineer who is not LSL maintainer more likely cannot do the same since there are no libs for it. And he or she will not implement LSL stuff from the scratch in Arduino environment I suppose. I would like to see the couple of device which implement it to change my mind.

```
I don't quite understand what you find difficult to 'automate' about LSL.
```

Quite often to read data from LSL you need to run 3rd party app to convert one stream to another, this app often is not well maintained(of course there are good apps as well).
It's not simple to automate the process of installation and launching of such app. You can not download this app from PYPI\Maven\Nuget automatically simply by adding it to the dependency list of your project and probably in real production environment you will need to bundle this app with your package and start it automatically somehow.

With device drivers(including BrainFlow) you just add it to dependency list.

```
LSL does not (usually) run on embedded hardware, so how could it possibly ever communicate directly with a device? Again, to talk to a device, as I stated above, one needs to go through the device driver.
```

And BrainFlow is a device driver which you run on PC, it sends command to device, LSL cannot do the same. 

In conclusion, it looks like you completely misunderstand how BrainFlow works and what was the main point of the post which you dont like.

I dont argue that LSL is good in time sync between different PCs, what I dont like is that it's often used in cases wo any networking and to work with single device(and seems like its the most common use case of LSL). For such cases IMO device drivers work better and provide much more functions to developers.

closed 2021-02-26 by Andrey1994

conorrussomanno, 2021-02-26:
At OpenBCI, we see significant value in both LSL and Brainflow and plan to continue to make our platform compatible with both. I think that the industry and open source BCI community would be better served if these tools were able to work in tandem with one another, rather than in opposition.

wjcroft, 2021-02-26:
I agree with both Conor and David that cooperation and synergy is what will move the field forward. As David mentions, I can see applications in which both systems would be used, each playing off their strengths. 

I would encourage Andrey to make some efforts to soften the 'tone' of his comparison blog post. It's not an "either or" situation. The need for synchronization between multiple lab devices will only increase as time passes, as the VR, AR, XR devices are showing.

Best regards, William Croft

matthijscox, 2021-02-26:
It's good to have this discussion here so we can weight the pros and cons publicly.

Linus Torvalds is known to swear profusely at developers for trying to be too clever. I think the blog represents Andrey cursing LSL for being too clever and accidentally leaving the world worse of. 

LSL is written with the best of intentions. But it's a pitfall for intellectuals to try to enforce new standards and protocols onto the world, as exemplified by this [XKCD comic](https://xkcd.com/927/). 

As a bio-sensor user and software developer I've not understood the need for another protocol for bio-sensors. I never used LSL, because as Andrey says I had to either find or write a 3rd party library to make the connection, after which I have the data. Why would I then continue to plumb the data into LSL?

In hindsight what we really needed was a userfriendly SDK. Fortunately BrainFlow fills that void now.

If it fills some need LSL will continue to attract new users. Someone may also write a BrainFlow to LSL adapter to integrate more easily with legacy applications. I personally will not look back though.

However I will follow this discussion with great interest.

wjcroft, 2021-02-26:
@matthijscox, hi. I'm not sure you are familiar with LSL, but it's major goal was (and is) extremely accurate time synchronization between experiments that utilize multiple sensor sampling devices. There are no other systems that can do this. LSL is in very wide usage in the neuroscience community for just this reason. 

Both LSL and Brainflow have valued positions in this field. They are NOT in 'competition', and it is short sighted to paint such a picture.

Regards, William Croft, OpenBCI

matthijscox, 2021-02-26:
@wjcroft very well, then it's good we're having this discussion to separate concerns between both open source projects.

I assumed the primary objective of LSL was to stream the data from any bio-sensor device. Secondary there were things like synchronization and centralized data collection. You now frame it the other way around.

Popularity aside, I want to know which tool is best for which task.

Andrey1994, 2021-02-26:
We can discuss the tone of some sentences or adoption of new best practices, use cases for BrainFlow and LSL, etc later on, here Dylan wrote that there is a "shitty misinformation" in this blog post.

So far I dont see any single argument for this opinion. Statements from blog post that LSL works on top of TCP are correct and you can not argue with it. Statements that there are no Arduino libraries for LSL are also correct as well as statements that in 99.9% of cases LSL needs another app which acts a a driver(check this https://github.com/brainflow-dev/brainflow/issues/246#issuecomment-786526556 for the full list)

Arguments like BrainFlow works the same way are wrong and I explained it above.
I would like to stop this conversation here at least until someone *provide real tech facts* to prove "misinformation".

And again I dont argue that LSL is good for time syncing between different PCs.

As I wrote above:

```
I dont argue that LSL is good in time sync between different PCs, what I dont like is that it's often used in cases wo any 

networking and to work with single device(and seems like its the most common use case of LSL). For such cases IMO device 
 
drivers work better and provide much more functions to developers.
```

wjcroft, 2021-02-26:
LSL excels at synchronization issues; because it is based on a network time protocol baseline standard (NTP) that provides a precise alignment of streams. And in actual neuroscience labs, experimental data is streaming from multiple sensors simultaneously and must be aligned; frequently to the sub-millisecond level. Please read some of the background if you are curious.

https://labstreaminglayer.readthedocs.io/info/intro.html

LSL and Brainflow are COMPLEMENTARY. The blog post could be reworded to emphasize synergy.

dmedine, 2021-02-26:
Thank you all for replying to my issue. I recognize that it was combattitive in tone, but I wanted to get people's attention because this is important stuff. I stand by my statements, but I think there is a little teensy bit of miscommunication happening. Which is fine, I am happy to let it go. If you wish to have an in depth technical discussion about how LSL works, I refer anyone interested to the LSL slack channel (https://github.com/sccn/labstreaminglayer, scroll down for the invite link).

I think BrainFlow is great and I very much hope that these two communities can find common ground. But, the OP needs to stop attacking LSL. As I said before, it makes BrainFlow look bad, too.

To be perfectly clear, what I really have a problem with in the blog post is this:

> This lack of communication with the device can also cause problems with timestamp synchronization, since LSL has no information about the transmission period from the device to the LSL application.

People have a hard time understanding the time sync functionality in LSL (which is understandable, it is very complicated and can fail if misused) and as someone who runs workshops and answers questions about this, I really don't need anyone spreading misinformation. That is shitty for me.

wjcroft, 2021-02-28:
@Andrey1994 hi. Can you please consider removing the sentence which David mentioned: "This lack of communication with the device can also cause problems with timestamp synchronization, since LSL has no information about the transmission period from the device to the LSL application."

You do not understand how LSL users in labs, carefully quantize the latency between their devices and the receiving process. This allows precise alignment with the NTP timing baseline. I previously showed you a week or so ago, an example of this. Which demonstrates how the UCSD lab quantified their BioSemi amp latency. This latency is then subtracted from the timestamps:

https://sccn.ucsd.edu/labinfo/equipment/latencies/biosemi.html

With this deliberate misstatement, you are propagating false information about LSL, and using it to denigrate the reputation of LSL in the larger community. This is not only unethical, but is hurting the reputation that Brainflow has for honesty and collaboration.

Mentioning @conorrussomanno @dmedine .

William Croft, OpenBCI

Andrey1994, 2021-02-28:
@wjcroft maybe you don't fully understand what I am talking about and whats written in this post, but ***article you posted proves my opinion in this blog post***

Could you answer please this simple question:

[image]

**Which approach from this image will provide better timestamps?**

It's obvious that first one will generate more accurate timestamps, and it's exactly whats written in blog post.  And it's not even the issue with LSL, it's a fundamental side effect of the most commonly used approach.

Now let's return to your article.

***Developers had to estimate this delay manually and they invested resources and their time into it.*** They would not need to do so if LSL streamer was working directly on the board. Also, they had to write this app carefully and ensure that this delay is stable.

It's a research lab and professional and expensive device. Huge amount of apps are worse and in plenty of cases there is no estimation for this latency. It makes resulting timestamp less accurate and blog posts says exactly this.

### Conclusion 
***The fact that you need to estimate this delay and the fact that it's not done in many apps proves this statement from the blog post.*** And it's not even the issue with LSL algorithms for time sync. So, as for me it's not a misstatement and because of this I am not going to remove it.

wjcroft, 2021-02-28:
Andrey, you understand nothing about LSL, nor NTP. NTP is too complex to put into 'devices'

Please just remove the sentence. Your hatred and petty judgements are hurting the open source community. Do you not have any ethics or morals? This entire conversation thread proves how vindictive you are.

**I'll underline that with this fact: your first post of this blog entry, used the URL:** 

https://brainflow.org/2021-02-05-kill-lsl/

"KILL LSL". In other words your intention all along was to do damage and "kill off" LSL from your point of view. This is not honest technical assessment. But instead vindictive and hugely biased opinion, to support your own products and diminish those of others. This is not how maturity works. Nor community spirit. Nor just plain honesty. I'd be rather ashamed of such behavior. If you were capable of actual feelings.

William Croft

Andrey1994, 2021-02-28:
"Andrey, you understand nothing about LSL, nor NTP. NTP is too complex to put into 'devices'"

For example neurosity does it directly from the board and I can provide other examples. NTP works perfectly on raspberry pi and there are arduino libraries to work with NTP.

wjcroft, 2021-02-28:
LSL works with commercial devices such as BioSemi, etc. In many cases these devices were created years ago and have no ability for complex firmware to incorporate NTP. You JUST DON'T GET IT. LSL is a general purpose solution for lab equipment with no ability to re-engineer the firmware. LSL is actually the CORRECT solution utilizing NTP for arbitrary devices.

Please just do the moral action that shows some integrity and community spirit. And remove the vindictive statement. Otherwise I'd have to conclude that Russians have a huge chip on their shoulders. Possibly equate you with Vladimir Putin.

Andrey1994, 2021-02-28:
This sentence removed. So, I hope now everybody should be satisfied.

@wjcroft you are blocked on this github org.

Instead tech discussion and facts you started to talk about politics here. It looks like a racism and not allowed.

dmedine, 2021-02-28:
Thank you @Andrey1994 for removing the offending line. I appreciate it very much. I also agree that the comments of @wjcroft concerning Russians and Putin are against the spirit of collaboration on github and are a violation of any acceptable code of conduct on an online forum. That was most unfortunate.

I do agree with him, though, that you seem to harbor some resentment towards LSL and that that resentment is not necessary. LSL does its job extremely well and it does not compete with BrainFlow. It does something that BrainFlow is not designed to do. For example, if you wish to record streams from two devices in order to study, say, EEG and eytracking, how would you synchronize them using BrainFlow? I guess you would have to call get_time_of_day in order to timestamp the data points. Then you could and compare the times the data are recorded. But, how would you account for jitter, clock drifts (no device operates at _exactly_ the sampling rate it advertises), dropped frames, etc? How would you timelock the two streams if the devices are streaming to different PCs? You would have a lot of work to do in order to make such a system, or you could simply use LSL which solves all those problems automatically. That is why I believe BrainFlow and LSL are complementary, and are not in competition with each other.

As for the picture you posted above, there are currently no opensource solutions I know of that can allow for sending LSL data directly from a hardware device. I know of a few commercial systems that do this, but they are closed. An LSL library for mbed or a similar bare-metal OS would be extremely useful, but it is not so easy. The problem is that LSL relies heavily on OS infrastructure for scheduling, memory management, and obtaining time information and the core library is extremely resource greedy. Thus, there is no simple way to create an embedded LSL solution. The core library needs to be surgically altered in order to do this.

Andrey1994, 2021-02-28:
"As for the picture you posted above, there are currently no opensource solutions I know of that can allow for sending LSL data directly from a hardware device. I know of a few commercial systems that do this, but they are closed. An LSL library for mbed or a similar bare-metal OS would be extremely useful, but it is not so easy. The problem is that LSL relies heavily on OS infrastructure for scheduling, memory management, and obtaining time information and the core library is extremely resource greedy. Thus, there is no simple way to create an embedded LSL solution. The core library needs to be surgically altered in order to do this."

You are right, I know all these facts, and I understand how LSL works pretty well and I know that it's not simple to port it. But I suppose you agree that if you run it directly from the board it would be much better. And it was the main point.

"How would you timelock the two streams if the devices are streaming to different PCs?"

There is no way to handle it in BrainFlow, I never made statements that BrainFlow supports it. And more likely will never do because its not designed for it.|
It should be done by library like LSL or using custom solution based on zeromq for example running on top of BrainFlow.

conorrussomanno, 2021-03-01:
I am really happy to see this conversation between LSL and BrainFlow moving in the right direction.

I want to address William Croft's remarks and say that he does not speak for OpenBCI in this matter. I am incredibly disappointed by his personal attack towards Andrey. 

While the tone and some of the technical claims of Andrey's original post greatly missed the mark, we all need to have the ability to disagree and debate the merits of different tools and technologies without burning bridges.

Andrey has been a valuable contributor to OpenBCI's software, and BrainFlow is a great addition to the open source BCI community.

Conor Russomanno
Co-founder & CEO @ OpenBCI