brainflow-dev/brainflow #256: Create Conan package

Andrey1994, 2021-03-07:
https://docs.conan.io/en/latest/getting_started.html

Andrey1994, 2021-03-08:
https://docs.conan.io/en/latest/creating_packages/existing_binaries.html

closed 2022-08-25 by Andrey1994

Andrey1994, 2023-07-28:
no, I just created it. I was going to upload it but there is too much bureaucracy to publish smth

stellarpower, 2023-07-28:
Understood. I am trying out [xmake](https://xmake.io/), it looks really useful and is going well so far. It can use packages directly from Conan, vcpkg, etc., but also has its own format - which may be less overhead for submitting to the repository. 

Currently Neuromore are just committing their dependencies into the repository as local files, and I was making the argument for moving to a package manager instead. So if that happens I may need to come back to this and take on publishing packages for Brainflow and a lot of other related biosignal libraries!

Andrey1994, 2023-07-28:
actually for neuromore its even worse, they've also changed source code and added their own build system so now its very difficult to update dependencies there. Instead they could add it as a cmake project or take precompiled binaries

stellarpower, 2023-07-28:
Yeah, had some issues with using plain Make too, so, I've suggested modernising it all to make it easier to work with. I am hoping xmake is going to be much nicer than CMake, and am suggesting using Earthly for reproducible cloud and local builds.