brainflow-dev/brainflow #265: Possible to have multiple boards connected simultaneously?

sjburwell, 2021-04-03:
Loving the BrainFlow library - makes R&D easy!

I am curious whether (through BrainFlow) it is possible to be connected to more than one board at once. E.g., I would like to collect EEG data from Notion2 as well as pulse and other peripheral data from a Cyton board. Does this sound feasible? 

Thanks in advance!
-Scott

Andrey1994, 2021-04-03:
Yes its possible and works fine. The only requirement - combination of board id and BrainFlowInputParams must be unique.

Andrey1994, 2021-04-03:
I have an example\test for synthetic board and one real board working at the same time https://github.com/brainflow-dev/brainflow/blob/master/tests/python/brainflow_multiboard_get_data.py

closed 2021-04-03 by Andrey1994

sjburwell, 2021-04-03:
Awesome! Thank you for the example code link, I will look into it!

maimonr, 2021-09-15:
Hi, I had a similar question, I hope this is the right place to ask it. Specifically, I have 2 of the same type of boards (both of them are Notion 2's), and I'd like to record from them simultaneously. I've tried the following:

```
board_id = BoardIds.NOTION_2_BOARD.value

params1 = BrainFlowInputParams()
params1.serial_number = str(serial_nums[0])
board1 = BoardShim(board_id, params1)

params2 = BrainFlowInputParams()
params2.serial_number = str(serial_nums[1])
board2 = BoardShim(board_id, params2)

board1.prepare_session()
board2.prepare_session()
```

But I get the following error when preparing the session for board2: "BrainFlowError: ANOTHER_BOARD_IS_CREATED_ERROR:16 unable to prepare streaming session"

I can successfully call prepare_session on either of the two boards independently, but not without closing the other's session first. Is there a workaround or something else I can try? Thanks!

Andrey1994, 2021-09-15:
answered in slack, cannot reproduce