brainflow-dev/brainflow #266: Major performance problems with latest brainflow

GrahamBriggs, 2021-04-04:
Some time between September 2020 and April 2 2021, something happened in the brainflow library code that degraded CPU usage performance by a factor of 2x.

Please see this video:  [https://youtu.be/IzZt7YjNEkc](https://youtu.be/IzZt7YjNEkc )

In the video, I start up my program using the brainflow libraries built on September 25 2020.  The program works fine and is using 15% of one core of a RPi 4B. 

I then stop the program, and replace the Sept 2020 libraries with libraries built from latest master on April 3 (I incorrectly say February in the video)

You will see that the exact same code, running with the new brainflow libraries is now using over 33% CPU.

Something happened in the library to cause a major performance problem, at least on the RPi platform.

To be fair, it is cool that the program can run at less than 10% overall CPU on a$35 computer. However, I used to be able to run the program also on a Pi zero at about 60% CPU (hot, but still some cycles to spare).  Using latest brainflow libraries, the program no longer runs effectively on Pi zero (Maxed out CPU constantly).

Any ideas?

Andrey1994, 2021-04-04:
Hi,

Could yo reproduce it using simple script\small samples?
It will be much more easier to profile, specially since I dont know which methods your app use

GrahamBriggs, 2021-04-05:
Andrey

I have attached a simple program file that shows this problem. 

Please also see this video demonstrating the issue.  
[https://youtu.be/IVA6sRNxhKI](https://youtu.be/IVA6sRNxhKI)

Running with September 2020 builds the sample is 1% CPU, running the latest builds, the sample is at 3% CPU.  

The video also shows my program:
Running at 10% using September library to read data from board
Running at 5% using September library in demo file mode (not reading data from board)
Running at 30% using latest library to read data from the board
Running at 5% using latest library in demo file mode

[brainflowPerformance.cpp.txt](https://github.com/brainflow-dev/brainflow/files/6255766/brainflowPerformance.cpp.txt)

Andrey1994, 2021-04-05:
it doesnt look like latest code... api was changed, now there is BrainFlowArray class

GrahamBriggs, 2021-04-05:
[image]

GrahamBriggs, 2021-04-05:
I am not sure what you mean about the API change. I have not changed my code and it is working with the latest libraries?

 I did build the instance of the brainflow from latest master commit 3113751 as you can see in the picture above.

Thanks
GB

Andrey1994, 2021-04-05:
Latest brainflow can not work with cpp file you shared since there were breaking changes in the api.

Take a look at this https://www.github.com/brainflow-dev/brainflow/tree/master/tests%2Fcpp%2Fget_data_demo%2Fsrc%2Fbrainflow_get_data.cpp

GrahamBriggs, 2021-04-05:
Andrey

The CPP file I shared was compiled using the old version of the library. As you saw in the video I ran the program, then swapped out the libraries with cut and paste.  If there was a breaking change I am shocked it ran at all, but you saw it run in the video.  Maybe this is not a valid test.  

I cooked up this test in a hurray on an older machine that I had not touched in a while (that still had old bf compiles on it).  I was motivated to do this test after I noticed performance problems on my main test machine.  The code I am running on my main machine is using brainflow libraries that are a branch from here:
[image]

The only modifications I have made from this  point are to add more logging in the cyton driver class.  This code runs at much higher CPU usage than the same code running on the older machine with the older libraries.  FWIW, this verifies that the change in performance happened some time between September and February 2.

I am confused when you say the code I built for my test is not the latest.  On April 3 I did fresh clone of master branch. You can see from the screen shot above I built from commit 3113751, which is the latest commit I see for master branch on GitHub:
[image]

 When I look at https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/inc/board.h I see the function declaration that I am calling, so this explains why my program ran, because this function is still in the code that I built:

[image]

Perhaps my cut and paste swap test is not a valid test here, but it does replicate the behavior I am seeing in my application. Code that runs at 10% using brainflow libraries built in September 2020 is now running at 30% using the libraries I built on the branch from the Feb2 commit. 

When you hear the sound of hoofs, think horses not zebras.

Andrey1994, 2021-04-05:
<< . If there was a breaking change I am shocked it ran at all, but you saw it run in the video

there is a static library for c++ binding and breaking change was there. It means that you need to rebuild your app, not only change .so files loaded in runtime

GrahamBriggs, 2021-04-05:
Andrey, this makes sense, and I accept that the test shown in the video was not valid. Unfortunately I don't have time to make another video to show you this effect using a proper build. For what is worth, the behavior seen in the video is the same behavior I see when I compare my app properly built with September libraries versus the app properly built with February version of the libraries at commit fc424b5.  Some time between September and February something changed in the code that had a major performance impact, at least on the RPi platform.

When I get some more time I will continue to investigate.  I encourage you to consider that there is actually a problem here and perhaps investigate with your own version of the libraries.  Using a simple sample and library code from September and February.

Thank you
GB

Andrey1994, 2021-04-05:
I can not reproduce it and in fact for some reason I see the opposite...

Simple test to check if there are any perf issues:
run it wo while loop, just start_stream and sleep in the main thread and check cpu usage, it will check if its related to data collection thread running in the background or not

if data collection thread is fine then its inside your main thread, and maybe not even related to brainflow part(the only brainflow call there is `get_board_data` I dont see any perf related changes in this method in git history)

Also, I compiled it both times directly on raspberry, wo any cross compilers\dockers\etc

Andrey1994, 2021-04-05:
In data collection thread the only change I see is a usage of indexes from json object instead hardcoded int constants.

Now its like 
```
            package[board_descr["accel_channels"][0].get<int> ()] = accel[0];
            package[board_descr["accel_channels"][1].get<int> ()] = accel[1];
            package[board_descr["accel_channels"][2].get<int> ()] = accel[2];
```

Before it was smth like:
```
            package[20] = accel[0];
            package[21] = accel[1];
            package[22] = accel[2];
```

I dont think that this change can lead to significant perf changes

GrahamBriggs, 2021-04-05:
Andrey

Thank you very much for taking the time for both the test and the code review to investigate this problem. I will go back and study my application more carefully, as well as make sure I am building the brainflow libraries properly and rebuilding my app accordingly.  You may close this issue if you wish.  If I continue to see problems I will try to narrow it down using a properly compiled test case and  I will get back to you.

Best regards, and thanks again for your effort on this great project

GB

closed 2021-04-05 by Andrey1994