brainflow-dev/brainflow #278: Major packet loss when multicasting at high sample rates

chrisjz, 2021-05-19:
I'm getting around 70-90% packet loss when multicasting with Cyton Wifi on Windows 10.

It's completely fine when it doesn't multicast, hitting around 1,000 samples per second. With Synthetic it's working fine with and without multicasting.

You can add this code between `data = board.get_board_data()` and `board.stop_stream()` in the [Python Get Data from a Board
](https://brainflow.readthedocs.io/en/stable/Examples.html#python-get-data-from-a-board) sample to check packet loss:

`samples_per_second = data.shape[1]`
`total_samples = samples_per_second / 10`
`packet_loss = "{:.2f}%".format((1 - (samples_per_second / (BoardShim.get_sampling_rate(args.board_id) * 10))) * 100)`
`print(f"Packet loss: {packet_loss} ({total_samples}/sps)")`

Run that script with Cyton Wifi using multicast in shell:
`get_data_from_board.py" --board-id 5 --ip-address 192.168.4.1 --ip-port 70000  --streamer-params streaming_board://225.1.1.1:6677`

Here's the result from capturing packets on Wireshark for 10 seconds with Cyton Wifi on multicast:

[image]

Andrey1994, 2021-05-19:
Hi,

Could you explain a little where exactly do you see lost packages? In the first process which reads data from a board or in the second one where you use streaming board?

Also,
```
packet_loss = "{:.2f}%".format((1 - (samples_per_second / (BoardShim.get_sampling_rate(args.board_id) * 10))) * 100)
```
this doesnt look like valid way to estimate package loss, you need to analyze package_num row instead

Andrey1994, 2021-05-19:
I would recommend to check buffer overflow as described here https://ref.onixs.biz/lost-multicast-packets-troubleshooting.html 

And probably to avoid buffer overflow it makes sense to run second process(with streaming board) before the main one(with multicast streamer)

I will try to set higher sampling rate for syntehtic board and reproduce it as well

chrisjz, 2021-05-19:
>  In the first process which reads data from a board or in the second one where you use streaming board?

It's in the first process which reads the data. Initially, I thought it was the second one using the streaming board, but realised it was indeed the first after looking into it deeper.

> this doesnt look like a valid way to estimate package loss, you need to analyze package_num row instead

Oh yes, sorry about that. I did export all those variations to csv files and those not hitting the intended sample rate were missing/skipping chunks of package number ranges. For example, with multicast on at 1,000 samples per second, consecutive rows did not have package_num from 0-255 but would skip a couple dozen roughly.

For example:
0,...
1,...
2,...
31,...
32,...
33,...
34,...
57,...
58,...
59,...
60,...
...

> I would recommend to check buffer overflow as described here https://ref.onixs.biz/lost-multicast-packets-troubleshooting.html
And tune it as described in this article, also probably to avoid buffer overflow it makes sense to run second process(with streaming board) before the main one(with multicast streamer)

Thank you for those pointers, I will try those out soon and report back.

Andrey1994, 2021-05-19:
"It's in the first process which reads the data. Initially, I thought it was the second one using the streaming board, but realised it was indeed the first after looking into it deeper"

If its in the 1st process it means that packages from device are lost and in this case I dont fully understand how multicast can affect it...

Also, to read data from cyton wifi we use tcp which ensure package delivery so I dont get how its possible

chrisjz, 2021-05-20:
> If it's in the 1st process it means that packages from device are lost and in this case I dont fully understand how multicast can affect it...

> Also, to read data from cyton wifi we use tcp which ensure package delivery so I dont get how its possible

I had similar thoughts, which makes this so odd. I've attached the network traffic captured by Wireshark for both Cyton Wifi and Synthetic which shows gaps in time on packets received with Cyton but consistent with Synthetic in case this helps.

[cyton-wifi-multicast-10-seconds.txt](https://github.com/brainflow-dev/brainflow/files/6513384/cyton-wifi-multicast-10-seconds.txt)
[synthetic-multicast-10-seconds.txt](https://github.com/brainflow-dev/brainflow/files/6513385/synthetic-multicast-10-seconds.txt)

> I would recommend to check buffer overflow as described here https://ref.onixs.biz/lost-multicast-packets-troubleshooting.html

I've run diagnostics on both network adaptor and OS buffer and nothing looks odd to me from the results. I didn't diagnose on app-level as the article didn't have instructions for Windows, and didn't tune either as it also just had instructions for Linux. I'll try to find ways to do this and report back.

These are the results when multicasting for 10 seconds with Cyton Wifi (before and after):

**Network adapter buffer**

[image]

**OS system network kernel buffers**

[image]

> And tune it as described in this article, also probably to avoid buffer overflow it makes sense to run second process(with streaming board) before the main one(with multicast streamer)

I tried this and no luck either.

Did you get a chance to run the synthetic board with a higher sample rate?

Andrey1994, 2021-05-20:
Could you run your tests for cyton wifi wo multicast?
If you lose packets in the first process multicast can not affect it at all

chrisjz, 2021-05-20:
Here's the result for cyton wifi wo multicast:

[cyton-wifi-no-multicast-10-seconds.txt](https://github.com/brainflow-dev/brainflow/files/6515179/cyton-wifi-no-multicast-10-seconds.txt)

Exporting that to csv file and looking at the results, it's hitting about 1,000 samples per second.

I re-ran cyton wifi with multicast and included both TCP and UDP log this time:

[cyton-wifi-multicast-10-seconds-incl-tcp.txt](https://github.com/brainflow-dev/brainflow/files/6515400/cyton-wifi-multicast-10-seconds-incl-tcp.txt)

Andrey1994, 2021-05-20:
Could you send stored data instead wireshart dumps?(use read write sample for reference or file streamer)
it will be easier to estimate sampling rate and package losses based on that

chrisjz, 2021-05-20:
Here it is with multicasting:
[cyton-wifi-multicast.txt](https://github.com/brainflow-dev/brainflow/files/6516274/cyton-wifi-multicast.txt)

Without:
[cyton-wifi.txt](https://github.com/brainflow-dev/brainflow/files/6516280/cyton-wifi.txt)

Andrey1994, 2021-05-20:
good,
here I dont see lost packages in the first file(no gaps in package num column) but I do see less datapoints for some reason. Its from the 1st process right?(not from streaming board)

chrisjz, 2021-05-20:
Yes, from the first process.

chrisjz, 2021-05-21:
I'll clarify just in case, it's the first process specified here:
https://brainflow.readthedocs.io/en/stable/SupportedBoards.html#streaming-board

```
params = BrainFlowInputParams()
params.ip_address = '192.168.4.1'
params.ip_port = 70000
board = BoardShim(BoardIds.CYTON_WIFI_BOARD.value, params)
board.prepare_session()
streaming_board = 'streaming_board://225.1.1.1:6677'
board.start_stream(450000, streaming_board)
```

Andrey1994, 2021-05-21:
Its weird since multicast streamer just calls `sendto` I didnt expect that this single call can be slow...
I can add it to another thread or add buffering for packages in multicast streamer and more likely it will solve the problem

chrisjz, 2021-05-21:
One of those approaches should work from what I’m understanding too researching this.

Andrey1994, 2021-05-23:
should be fixed in this PR

PR #280 (2021-05-23): buffering for multicast streamer
Speed up multicast streamer using buffering and another thread to close #278
closed 2021-05-23 by Andrey1994