brainflow-dev/brainflow #298: Timestamp is the same for all records in get_board_data output

nastaran62, 2021-06-24:
I have a cyton-daisy Openbci board and I used get_board_data to read data from the buffer. It seems that each time that I call this method, all records have the same timestamp! 
I need to read data sample by sample to apply some formatting on each sample. The following is part of my code for reading data. In the output list, timestamp is the same for each two, three, or ... records. It seems that each time that I call this method, all records in the buffer have the same timestamp! 

data_list = []
while True:
            if terminate is True:  # A flag which is changing by another thread
                break
            data = self._board.get_board_data()
           # In some loops buffer still is empty
            if np.array(data).shape[1] is not 0:
                data_list.extend(list(np.transpose(data)))

Andrey1994, 2021-06-24:
Hi, its not the same, its changing over time. You can run basic sample from the docs to confirm that. Some of them can be close because of https://docs.openbci.com/docs/10Troubleshooting/FTDI_Fix_Windows

closed 2021-06-24 by Andrey1994

nastaran62, 2021-06-25:
The following are the last columns of my output. The second column from the end is timestamp and some of them are the same. 

112.5 | 24 | 0 | 0 | 0 | 1624421714.45235 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45236 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45236 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45236 | 0
112.5 | 40 | 0 | 0 | 0 | 1624421714.45237 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45237 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45237 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.45237 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.51248 | 0
112.5 | 40 | 0 | 0 | 0 | 1624421714.51248 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.51248 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.51249 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.51249 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.51249 | 0
112.5 | 40 | 0 | 0 | 0 | 1624421714.51249 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.57277 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.57278 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.57279 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.57279 | 0
112.5 | 32 | 0 | 0 | 0 | 1624421714.57279 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.5728 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.5728 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.57282 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.63309 | 0
112.5 | 32 | 0 | 0 | 0 | 1624421714.6331 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.6331 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.6331 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.6331 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.63311 | 0
112.5 | 32 | 0 | 0 | 0 | 1624421714.63311 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.69247 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.69247 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.69248 | 0
0 | 0 | 0 | 0 | 0 | 1624421714.69248 | 0
112.5 | 16 | 0 | 0 | 0 | 1624421714.69248 | 0

</body>

</html>

Andrey1994, 2021-06-25:
Its a side effect of ftdi buffering which you can disable using the link I sent. Because of buffering multiple samples read in a row wo delay

nastaran62, 2021-06-25:
Thank you very much.

nastaran62, 2021-07-05:
Hi,
When I change timestamp from 16 to 1, timestamps are not the same for various samples and it's ok. I'm a Linux user and for changing FTDI buffer I need sudo access. It means I cannot run the code in systems that I don't have sudo access. Also, each time I connect dongle to USB port, I should again set the FTDI buffer. Do you have any solution for this issue?

Andrey1994, 2021-07-05:
As far as I know for linux there is no such buffering, but I am not sure. Its a question to OpenBCI guys(check openbci forum), such driver settings are not part of BrainFlow.
Alternative solution - you can add your own timestamps based on sampling rate if you want, I dont do it in BrainFlow on purpose to represent real data streaming

nastaran62, 2021-07-05:
Thanks a lot. I will. 
I'm already using the pyOpenBCI library in python as well and don't have a similar issue because It provides reading data sample by sample. I couldn't find any module in brainflow for reading data record by record, If I can read data record by record I can handle it myself.

Andrey1994, 2021-07-05:
it reads data sample by sample in another thread and assignes timestamps to each sample as soon as its received.

nastaran62, 2021-07-06:
If it is doing it I will not have this issue. Which module read data sample by sample?  I am using get_board_data.
Does it mean if I have similar timestamps, several samples arrived at the same time?
get_board_data directly flush FTDI buffer?

Andrey1994, 2021-07-06:
"Does it mean if I have similar timestamps, several samples arrived at the same time?"

Yes, because they are buffered by ftdi driver

Andrey1994, 2021-07-06:
You dont understand the diff between ftdi driver buffering and storing data in a ring buffer in brainflow