brainflow-dev/brainflow #300: How to use Brainflow with custom hardware

ut33, 2021-06-27:
I've working on a custom signal acquisition board and I would like to use Brainflow to stream data from python, what should I do?
Thanks in advance

Andrey1994, 2021-06-27:
Hi, here is a guide how to add new boards https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html#instructions-to-add-new-boards-to-brainflow 
If you have questions about adding feel free to ask

closed 2021-06-27 by Andrey1994

ut33, 2021-10-14:
Hi,
My board has a sampling rate of 1000 sps and 32 channels, so every
millisecond it gets 32 data points and every second it has 32000 data
points. Will brainflow be able to plot and record this data in real time?
Best Regards,
Vini

ut33, 2021-10-14:
> Hi, brainflow doesnt plot anything, its an sdk, not gui app.
>
> Its written in c/c++ so you can apply all possible optimizations in
> implementation of your device

Thank you for your response.
My board has a sampling rate of 1000 sps and 32 channels, so every
millisecond it gets 32 data points and every second it has 32000 data
points. Will brainflow be able to record this data in real time?
Best Regards,
Vini