brainflow-dev/brainflow #326: [board_logger] [error] failed to start stream2

wangzhengtao1993, 2021-08-15:
HI @Andrey1994 ,
I’m writing a Qt App in C++. It can prepare the session, but it cannot start streaming. I’ve tried the python code, and it works fine, so I think it is not a bug from the hardware. I wonder if anyone has this problem. I would appreciate it if you could help me solve this problem.

Brainflow version: 4.5.0
OS: Windows 10
IDE: VS2019

I did the following steps:
1. I used cmake3.20.5 to generate a inc and lib folder
2. Add a new Qt Widgets Application project
3. Project property->VC++ Directories->Include Directories:D:\code2021\QtVS\cmake\brainflow_prefix-4.3.1\inc;$(IncludePath)
5. Project property->C/C++->Code Generation->Runtime Library: Multi-threaded Debug (/MTd)
6. Project property->Linker->General->Additional Library Directories: D:\code2021\QtVS\cmake\brainflow_prefix-.3.1\lib;%(AdditionalLibraryDirectories)
7. Project property-> Linker->Input->Additional Dependencies: BoardController.lib;BrainBitLib.lib;Brainflow.lib;DataHandler.lib;DSPFilters.lib;GanglionLib.lib;MLModule.lib;MuseLib.lib;WaveLib.lib;%(AdditionalDependencies)

Code:
```
#include "EMGWidget.h"
#include "board_shim.h"

EMGWidget::EMGWidget(QWidget* parent)
	: QWidget(parent)
{
	ui.setupUi(this);
	connectBoard();
}

void EMGWidget::connectBoard()
{
	BoardShim::enable_dev_board_logger();
	struct BrainFlowInputParams params;
	params.ip_address = "192.168.1.105";
	params.ip_port = 8888;
	int res = 0;
	int board_id = 6;
	BoardShim* board = new BoardShim(board_id, params);

	try
	{
		board->prepare_session();
		qDebug() << board->is_prepared();
		board->start_stream();
	}
	catch (const BrainFlowException& err)
	{
		BoardShim::log_message((int)LogLevels::LEVEL_ERROR, err.what());
		res = err.exit_code;
		if (board->is_prepared())
		{
			board->release_session();
		}
	}
}
```

Logs:
```
[2021-08-15 12:19:14.883] [board_logger] [info] incoming json: {"file":"","ip_address":"192.168.1.105","ip_port":8888,"ip_protocol":0,"mac_address":"","other_info":"","serial_number":"","serial_port":"","timeout":0}
[2021-08-15 12:19:14.884] [board_logger] [info] use 10 as http timeout
[2021-08-15 12:19:14.891] [board_logger] [info] local ip addr is 192.168.1.104
[2021-08-15 12:19:16.980] [board_logger] [info] configuration string {"delimiter":true,"ip":"192.168.1.104","latency":10000,"output":"raw","port":8888}
[2021-08-15 12:19:19.075] [board_logger] [info] command string {"command":"~4"}
[2021-08-15 12:19:21.209] [board_logger] [warning] If you change gain you may need to rescale EXG data
[2021-08-15 12:19:21.210] [board_logger] [info] command string {"command":"d"}
[2021-08-15 12:19:23.600] [board_logger] [warning] If you change gain you may need to rescale EXG data
true
[2021-08-15 12:19:23.758] [board_logger] [error] failed to start stream2
```

wangzhengtao1993, 2021-08-15:
I tried an older version, 4.3.1, and the code sample works.

Andrey1994, 2021-08-15:
Hi, could you confirm that it's an issue with the latest version from the master branch?
I can only recommend adding more log messages and rebuild or use visual studio\qt creator debugger...

Here is an exception which you see https://github.com/brainflow-dev/brainflow/blob/master/cpp-package/src/board_shim.cpp#L121 

`res==2` means `UNABLE_TO_OPEN_PORT_ERROR` from  https://github.com/brainflow-dev/brainflow/blob/master/src/utils/inc/brainflow_constants.h#L7 

But `start_stream` method for this board doesn't return anything like that https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/openbci/openbci_wifi_shield_board.cpp#L222

wangzhengtao1993, 2021-08-15:
I just redo the steps from the beginning using 

> brainflow-4.5.0.zip

And it works in **Release configuration**, not in **Debug configuration**. So wired.

I think it might be related to the Runtime Library? The default settings would bring errors like
```
1>Brainflow.lib(board_shim.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in EMGWidget.obj
1>Brainflow.lib(board_shim.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MTd_StaticDebug' doesn't match value 'MT_StaticRelease' in EMGWidget.obj
```
Only **Multi-threaded Debug (/MTd)** works. I should learn more about those things.

Yes, the start_stream doesn’t return anything. I can get data using get_current_board_data and print. 

Anyway, it works now. You guys did a great job, thank you.

Andrey1994, 2021-08-15:
For debug configuration in QT you need to compile brainflow using debug mode also. If you are using build.py you need to prove `--debug` argument

closed 2021-08-15 by Andrey1994