brainflow-dev/brainflow #327: Add swift binding

Andrey1994, 2021-08-16:
Use other languages(java, c#, python) as an API reference.

Useful link: https://stackoverflow.com/questions/34669958/swift-how-to-call-a-c-function-loaded-from-a-dylib

ScottThomasMiller, 2021-08-19:
I am trying to create a bridging header for Swift 5 and Xcode 12.5.1.  Must my Swift code instantiate the BoardShim object, or can it just call prepare_session() and start_stream() directly?

Andrey1994, 2021-08-19:
The idea of bindings is to provide high-level API which is easier to use than plain C method exposed from low-level library. So, we created BoardShim class for each binding. Users are expected to work with BoardShim methods instead calling low-level prepare_session and other methods directly.

Example from Java:(https://github.com/brainflow-dev/brainflow/blob/master/java-package/brainflow/src/main/java/brainflow/BoardShim.java)

```
    public void prepare_session () throws BrainFlowError
    {
        int ec = instance.prepare_session (board_id, input_json);
        if (ec != ExitCode.STATUS_OK.get_code ())
        {
            throw new BrainFlowError ("Error in prepare_session", ec);
        }
    }
```

Andrey1994, 2021-08-19:
So I would say let's create BoardShim, DataFilter, and other classes like in existing bindings

Andrey1994, 2021-08-19:
I dont know swift but why do we need a bridging header at all?(and what is it?)

I think it can look like this(pseudocode inspired from python binding):

```
class BoardShimDynLibLoader {
    handler = nil;
    static func get_handler() {
        if (handler == nil)
            handler = dlopen("libBoardContoller.dylib", RTLD_NOW)
        return handler;
}

class BoardShim {
    func prepare_session() {
        typealias prepare_session = @convention(c) (CInt, Cchar *) -> CInt
        handler = BoardShimDynLibLoader.get_handler()
        var prepare_session = dlsym(handle, "prepare_session")
        if (prepare_session != nil) {
            let f = unsafeBitCast(sym, prepare_session .self)
            let result = f(board_id, params)
       }
    }    
}
```

Idea is to create a singleton that will call `dlopen` only once and find addresses of required functions manually.
Is it worse or better?(I don't know swift best practices)

ScottThomasMiller, 2021-08-19:
The bridging header tells Swift how to resolve the object names in your Swift code.  Without it you get errors such as, "Cannot find 'prepare_stream' in scope".  It's typically a .h file which simply includes other .h files.  You have to register it in the Xcode build settings, so that Swift knows where to look.

Your example is similar to this: https://stackoverflow.com/questions/35229149/interacting-with-c-classes-from-swift

I should be able to implement something like the that, because I already have all of BrainFlow's C++ headers.

I am still learning Swift, but I think that your idea is basically a wrapper, and wrappers are very "Swifty".

ScottThomasMiller, 2021-08-23:
I managed to call the low-level functions directly, such as prepare_session() and get_eeg_channels(), but so far I am unable to instantiate BoardShim().  I need some help.  Everything I've found so far, I can't get working.

Andrey1994, 2021-08-23:
You should not call methods from C++ BoardShim class. This class even doesn't exist in low-level API(low-level API is for bindings, high-level API is for users).

Instead, you need to create a new class in Swift and call it BoardShim, inside this class you need to wrap methods from low-level API. Take a look at the cpp-package for reference. https://github.com/brainflow-dev/brainflow/blob/master/cpp-package/src/board_shim.cpp#L104 Here method `is_prepared` from BoardShim class wraps method `is_prepared` from low-level API, you need to do the same

ScottThomasMiller, 2021-08-23:
OK I think I finally understand: you want me to write BoardShim.swift modeled after board_shim.cpp.  That seems completely doable.  I will also have to write swift bindings for all of BoardShim's dependencies, such as BrainFlowException, BrainFlowArray, BrainFlowInputParams, etc.  And nowhere in all of those bindings will I need to instantiate any C++ classes, correct?

Andrey1994, 2021-08-23:
Correct, but probably you dont need BrainFlowArray(its specific for c++ for better perf, you can do the same as in java for example). Also, there are two more classes: DataFilter and MLModule, its the same there.(I can help and do it for some of them)

Andrey1994, 2021-08-23:
Also, you need to duplicate enums like BoardIds, etc. All of that is pretty small and simple except main classes(BoardShim and DataFilter)

ScottThomasMiller, 2021-08-24:
Here's what I have so far:

struct BrainFlowInputParams: Encodable
{
    var serial_port: String = ""
    var mac_address: String = ""
    var ip_address: String = ""
    var ip_port: Int = 0
    var ip_protocol: Int = 0
    var other_info: String = ""
    var timeout: Int = 0
    var serial_number: String = ""
    var file: String = ""
    
    // Convert the BrainFlowInputParams struct to a JSON string:
    func json() -> String {
        let encoder = JSONEncoder()
        encoder.outputFormatting = .prettyPrinted.union(.withoutEscapingSlashes)

        guard let jsonParams = try? encoder.encode(self) else {
            print("Cannot convert BrainFlowInputParams to JSON")
            return ""
        }
        
        let stringParams = String(data: jsonParams, encoding: .utf8)
        return stringParams!
    }
}

enum BrainFlowExitCodes : Int32
{
    case STATUS_OK = 0
    case PORT_ALREADY_OPEN_ERROR = 1
    case UNABLE_TO_OPEN_PORT_ERROR = 2
    case SET_PORT_ERROR = 3
    case BOARD_WRITE_ERROR = 4
    case INCOMMING_MSG_ERROR = 5
    case INITIAL_MSG_ERROR = 6
    case BOARD_NOT_READY_ERROR = 7
    case STREAM_ALREADY_RUN_ERROR = 8
    case INVALID_BUFFER_SIZE_ERROR = 9
    case STREAM_THREAD_ERROR = 10
    case STREAM_THREAD_IS_NOT_RUNNING = 11
    case EMPTY_BUFFER_ERROR = 12
    case INVALID_ARGUMENTS_ERROR = 13
    case UNSUPPORTED_BOARD_ERROR = 14
    case BOARD_NOT_CREATED_ERROR = 15
    case ANOTHER_BOARD_IS_CREATED_ERROR = 16
    case GENERAL_ERROR = 17
    case SYNC_TIMEOUT_ERROR = 18
    case JSON_NOT_FOUND_ERROR = 19
    case NO_SUCH_DATA_IN_JSON_ERROR = 20
    case CLASSIFIER_IS_NOT_PREPARED_ERROR = 21
    case ANOTHER_CLASSIFIER_IS_PREPARED_ERROR = 22
    case UNSUPPORTED_CLASSIFIER_AND_METRIC_COMBINATION_ERROR = 23
}

enum BoardIds : Int32
{
    case PLAYBACK_FILE_BOARD = -3
    case STREAMING_BOARD = -2
    case SYNTHETIC_BOARD = -1
    case CYTON_BOARD = 0
    case GANGLION_BOARD = 1
    case CYTON_DAISY_BOARD = 2
    case GALEA_BOARD = 3
    case GANGLION_WIFI_BOARD = 4
    case CYTON_WIFI_BOARD = 5
    case CYTON_DAISY_WIFI_BOARD = 6
    case BRAINBIT_BOARD = 7
    case UNICORN_BOARD = 8
    case CALLIBRI_EEG_BOARD = 9
    case CALLIBRI_EMG_BOARD = 10
    case ALLIBRI_ECG_BOARD = 11
    case FASCIA_BOARD = 12
    case NOTION_1_BOARD = 13
    case NOTION_2_BOARD = 14
    case IRONBCI_BOARD = 15
    case GFORCE_PRO_BOARD = 16
    case FREEEEG32_BOARD = 17
    case BRAINBIT_BLED_BOARD = 18
    case GFORCE_DUAL_BOARD = 19
    case GALEA_SERIAL_BOARD = 20
    case MUSE_S_BLED_BOARD = 21
    case MUSE_2_BLED_BOARD = 22
    case CROWN_BOARD = 23
    case ANT_NEURO_EE_410_BOARD = 24
    case ANT_NEURO_EE_411_BOARD = 25
    case ANT_NEURO_EE_430_BOARD = 26
    case ANT_NEURO_EE_211_BOARD = 27
    case ANT_NEURO_EE_212_BOARD = 28
    case ANT_NEURO_EE_213_BOARD = 29
    case ANT_NEURO_EE_214_BOARD = 30
    case ANT_NEURO_EE_215_BOARD = 31
    case ANT_NEURO_EE_221_BOARD = 32
    case ANT_NEURO_EE_222_BOARD = 33
    case ANT_NEURO_EE_223_BOARD = 34
    case ANT_NEURO_EE_224_BOARD = 35
    case ANT_NEURO_EE_225_BOARD = 36
    case ENOPHONE_BOARD = 37
    // use it to iterate
    // (not yet sure how to do this in Swift)
    //case FIRST = PLAYBACK_FILE_BOARD
    //case LAST = ENOPHONE_BOARD
}

class BrainFlowException: Error {
    var message: String
    var errorCode: Int32
    
    init(_ errorMessage: String, _ code: Int32) {
        message = errorMessage
        errorCode = code
    }
}

struct BoardShim {
    let boardId: BoardIds
    let params: BrainFlowInputParams
    let jsonParams: String
    
    init (_ boardId: BoardIds, _ params: BrainFlowInputParams) {
        self.boardId = boardId
        self.params = params
        self.jsonParams = params.json()
    }
    
    func prepareSession() throws {
        let result = prepare_session(boardId.rawValue, jsonParams)
        let exitCode = BrainFlowExitCodes(rawValue: result)
        if exitCode != BrainFlowExitCodes.STATUS_OK {
            throw BrainFlowException("failed to prepare session", result)
        }
    }
    
    func isPrepared () throws -> Bool  {
        var intPrepared: Int32 = 0
        let result = is_prepared (&intPrepared, boardId.rawValue, jsonParams)
        let exitCode = BrainFlowExitCodes(rawValue: result)
        if exitCode != BrainFlowExitCodes.STATUS_OK {
            throw BrainFlowException ("failed to check session", result)
        }
        guard let boolPrepared = Bool(exactly: NSNumber(value: intPrepared)) else {
            throw BrainFlowException ("is_prepared returned non-boolean", intPrepared)
        }
        return boolPrepared
    }

    func releaseSession () throws {
        let result = release_session (boardId.rawValue, jsonParams)
        let exitCode = BrainFlowExitCodes(rawValue: result)
        if exitCode != BrainFlowExitCodes.STATUS_OK {
            throw BrainFlowException ("failed to release session", result);
        }
    }

}

func testBrainFlow() {
    let params = BrainFlowInputParams(serial_port: "/dev/cu.usbserial-DM0258EJ")
    let board = BoardShim(BoardIds.CYTON_DAISY_BOARD, params)
    do {
        try board.prepareSession()
        try board.isPrepared() ? print("open") : print("closed")
        
        try board.releaseSession()
        try board.isPrepared() ? print("open") : print("closed")
    } catch {
        print("Error: \(error)")
    }
}

ScottThomasMiller, 2021-08-24:
Here's what I have so far: https://github.com/ScottThomasMiller/ScottsML/tree/main/Swift/BrainFlowBindings

ScottThomasMiller, 2021-08-24:
Thanks.  Swift doesn't like binding its String type to "char *" parameters in its bridging headers, so I'm making them all "const char *" in board_controller.h instead.

Also I think Swift will automatically make BoardIds iterable if I add the CaseIterable protocol to its type definition.  I'll try that shortly.

Andrey1994, 2021-08-24:
"I'm making them all "const char *" in board_controller.h instead."

Its not ok, some other bindings don't like `const char *`. Lets find a way to keep board_controller.h as is(add type casting somewhere on swift side)

Andrey1994, 2021-08-24:
I cannot say that I fully understand what exactly they are doing here https://forums.swift.org/t/how-to-pass-swift-string-to-c-function-in-swift-5/28416 but seems like its doable with `char *` in C header

ScottThomasMiller, 2021-08-24:
Understood.  I'll take a look.

ScottThomasMiller, 2021-08-24:
I reverted board_controller.h to its original state.  For the board parameter string I am now converting the JSON string inside BoardShim.init() like so:

self.cParams = UnsafeMutablePointer<CChar>(mutating: jsonParams)

And then I pass cParams into the low-level functions.  It seems to be working OK.

Swift gives me a warning about creating a dangling pointer, at the preceding line of code.  From what I've read, if the low-level functions never modify or destroy the JSON parameters, then we can safely ignore the warning.

ScottThomasMiller, 2021-08-25:
I need some more help, this time with BrainFlowArray.  My C++ is weak.  I cannot translate BrainFlowArray into Swift all by myself.  From what little I do understand of it, BrainFlowArray provides various matrix operations, including reshaping the input buffer in a sort of numpy fashion. 

For the app I am building, all I need is a 2D matrix of doubles, so I will focus on that initially.  For testing purposes, to make sure I reshape the buffer correctly, what do you suggest?  Is there a simulator mode I can use to compare results against?

Andrey1994, 2021-08-25:
You dont need to port BrainFlowArray to Swift, its numpy-like optimization specific for C++. Key idea there is memory layout(2d array is 1d array in fact) 

You can do the same as in java https://github.com/brainflow-dev/brainflow/blob/master/java-package/brainflow/src/main/java/brainflow/BoardShim.java#L779 there its just standard 2d array

Andrey1994, 2021-08-25:
More info if you are interested: https://brainflow.org/2021-03-09-new-major-version/

Andrey1994, 2021-08-25:
"For the app I am building, all I need is a 2D matrix of doubles, so I will focus on that initially. For testing purposes, to make sure I reshape the buffer correctly, what do you suggest?"

Run it using synthetic board and check that rows are correct(first row is package num, etc)

ScottThomasMiller, 2021-08-29:
Your "Java Signal Filtering" example contains the following line of code:

                    DataFilter.perform_lowpass (data[eeg_channels[i]], BoardShim.get_sampling_rate (board_id), 20.0, 4,
                            FilterTypes.BESSEL.get_code (), 0.0);

The example fetches 30 samples.  In the preceding line of code, does data[eeg_channels[i]] represent a 30-element vector of doubles for the ith channel?

Andrey1994, 2021-08-29:
yes

ScottThomasMiller, 2021-08-29:
Follow-up question: in the preceding example, data is a 2D array of doubles, which is why you are iterating over each channel, calling the filter once for each channel.  Is it OK if I instead use a slice of the 1D buffer, before it's been reshaped, to call the filter only once for all channels?  

For example, if I have 10 samples and 16 channels, then the first 10 elements of the 1D buffer are package IDs, and the next 160 elements are EEG channel data.  Is it OK to call the filter just once using data[10...169]?

Andrey1994, 2021-08-29:
No, it's not ok.

These 2 APIs are independent and BoardShim should return 2d array while existing methods for signal processing operate on a 1d array. Also, from the user's point of view, it allows you to apply different filters for different channels.

ScottThomasMiller, 2021-08-29:
I see what you mean, and indeed I plan to provide matching functionality in my bindings.  But what I would like to also do is to overload get_board_data() with second function which returns the 1D buffer.  Users will still be able to call the original function and get the 2D array.  The two APIs will remain independent.

My question is more about whether the filter will still return the correct values if I pass in all channels at once.

Andrey1994, 2021-08-29:
For correct implementation of 1d array it should be like in C++ binding, we can add it later on. For beginning, I think we should make it like in java.

As is filter function will not understand that its multiple arrays joined together in 1d  array, it will treat it as a single 1d array and will generate wrong result. To fix it method signature for filtering should be changed. 

I was going to implement smth like that before https://github.com/brainflow-dev/brainflow/issues/108 but have no time and don't fully understand how to make it the same for all bindings

Andrey1994, 2021-09-05:
Hi, how is it going? If you need some help you can create a WIP PR

ScottThomasMiller, 2021-09-05:
It's going well.  BoardShim is completely coded, but not yet 100% tested.  I'm working on building a unit test module for it.  DataFilter is only about 25% coded.

ScottThomasMiller, 2021-09-06:
Actually I do have one remaining question about BoardShim: in your BoardShim.java, the get_num_rows binding calls instance.get_num_rows, but the get_board_data and get_current_board_data bindings call BoardShim.get_num_rows.  Why is that?

Andrey1994, 2021-09-06:
`instance.get_num_rows` is a method from low-level library(BoardController) and this method is called from `BoardShim.get_num_rows`. Later on in user-code and\or in other methods from `BoardShim` class you can call `BoardShim.get_num_rows`  while `instance` object is private and not for end users

ScottThomasMiller, 2021-09-09:
I'm not yet sure how to do the same in Swift, nor am I certain it's necessary.  

I am ready to push BoardShim.swift and its dependencies.  Please give me instructions for committing and pushing.

Andrey1994, 2021-09-09:
<<I'm not yet sure how to do the same in Swift, nor am I certain it's necessary.

This stuff with `instance` in java is kinda bridging header in swift. It describes the interface of BoardController.dll and allows you to call C\C++ methods. You don't need that.

<<I am ready to push BoardShim.swift and its dependencies. Please give me instructions for committing and pushing.

Create WIP PR from non-master branch of your fork and if you see a checkbox like "allow edit to maintainers" click on it. So, I will be able to make changes and help you a little. Make sure that swift binding matches the common structure in terms of folder and file names. Would be good to add a few tests to CI pipelines but it can be done later on

ScottThomasMiller, 2021-09-09:
In my fork should I duplicate ./brainflow/settings.xml and ./brainflow/pom.xml?

Andrey1994, 2021-09-09:
no, these files are specific for maven packaging(like setup.py in python)

ScottThomasMiller, 2021-09-09:
I created my fork and within it a branch named swift-bindings.  I cloned that branch to my laptop, added my .swift files, committed and pushed them, and now am ready to create my pull request from .  How do I make it WIP?  

Here's a screenshot:

[image]

Andrey1994, 2021-09-09:
Green Button "Create pull Request" click on triangle close to it

ScottThomasMiller, 2021-09-09:
"Create draft pull request"?

Andrey1994, 2021-09-09:
yes

ScottThomasMiller, 2021-09-11:
Because of the Swift scope rules, I had to move the logger functions of DataFilter inside its type definition.  Otherwise it would have caused name collisions with the analogous BoardShim functions.  For consistency I should probably do the same for the BoardShim logger functions.  What do you think?

Andrey1994, 2021-09-11:
I think all methods like `get_exg_channels` ,`set_log_level`, etc should be static class methods instead of global functions. This way it will be consistent with other bindings

Andrey1994, 2021-09-11:
In fact there should not be any global functions, only class methods and static class methods

Andrey1994, 2021-09-11:
And lets move this discussion to PR comments

closed 2026-09-13 by Andrey1994