brainflow-dev/brainflow #328: Bluetooth library improvements: add autodiscovery of paired devices for linux

Andrey1994, 2021-08-16:


lvntky, 2021-08-30:
Hi @Andrey1994 

I would like to take this issue, but it will be only the second open-source project that will I contributed to. So if you can inform me about the issue I think I can handle this.

Andrey1994, 2021-08-30:
Hi @lvntky, sure, and thanks for working on it.

There is a helper library for Bluetooth connection inside brainflow. It can connect to paired devices by MAC addr or it can try to discover device by part of the name. Code for it can be found here:

* for Win https://github.com/brainflow-dev/brainflow/blob/master/src/utils/bluetooth/socket_bluetooth_win.cpp#L138
* for Linux it currently returns UNIMPLEMENTED_ERROR https://github.com/brainflow-dev/brainflow/blob/master/src/utils/bluetooth/socket_bluetooth_linux.cpp#L109 
* and example where its called from https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/enophone/enophone.cpp#L329 

In this task, we want to add implementation for Linux. A few links to help https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html#code-style https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#compilation-of-core-module-and-c-binding in build.py you need to provide `--bluetooth` argument

lvntky, 2021-08-31:
okay @Andrey1994 thank you for the information :) I'm onto it.

Andrey1994, 2021-08-31:
I think the easiest way for you to test it would be:

* connect any Bluetooth device to your PC and get its name
* temp change name prefix for enophone https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/enophone/enophone.cpp#L341 to match prefix for your device
* make your changes about discovery, build, install and run brainflow_get_data.py example for enophone device wo mac address provided and check that your device found
* this example of course will fail later on since its another device but it doesn't matter