brainflow-dev/brainflow #336: Add emulator for Bluetooth devices

Andrey1994, 2021-08-29:


abhishekchopra0907, 2021-08-29:
Hi @Andrey1994 

Can I take up this issue, this is my first open-source issue, I couldn't completely comprehend the issue, can you please elucidate it a bit more. Thank you

Andrey1994, 2021-08-29:
Hi @abhishekchopra0907!

Sure, you can take this one, as well as any other opened issue here)
In CI pipelines we run tests and runners deployed in the cloud. It means that we cannot connect real hardware to run tests. 

To workaround it we try to develop emulators of real devices which work in the VM\Container deployed in the cloud.

You can take a look at:

* https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html#brainflow-emulator 
* https://github.com/brainflow-dev/brainflow/tree/master/emulator
* https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/run_unix.yml#L144

We did it for serial port, and for socket connection. Would be good to do the same for Bluetooth(maybe its enough to implement RFCOMM Server)

Take a look at enophone device implementation, there is "client" code to test.

This issue can be a little complicated, and I am not sure that rfcomm server will work fine.

closed 2022-06-02 by Andrey1994