brainflow-dev/brainflow #341: Is Neurosity Crown streaming at wrong sampling interval? I am clocking 4.0 ms (1000 ms / 250 Hz) instead of 3.90625 ms (1000 ms / 256 Hz).

sjburwell, 2021-09-04:
**Describe the bug**
I am testing the Neurosity Crown for recording EEG/ERP data in [eeg-notebooks](https://github.com/NeuroTechX/eeg-notebooks), and I believe there is an issue that stems from `brainflow` or possibly the Neurosity API. Specifically, when I inspect the timestamps associated with each streamed data sample, their modal first difference (i.e., sampling interval) is 4.0 instead of 3.90625 ms. The former sampling interval would be expected for the Neurosity Notion2 (which has a sampling rate of 250 Hz), but not the Crown (which has a sampling rate of 256 Hz). Thoughts on what might be happening? @alexcastillo?

**To Reproduce**
Output data streamed from the Crown with timestamps to a CSV file, inspect the timestamp data's first difference (row-wise). There is substantial variability (an OSC issue?), but the modal sampling interval is 4.0 when it should be 3.90625 ms.

**Info (please complete the following information):**
 - Board Id: Crown
 - OS running on your PC: Mac
 - Programming Language and its version: Python

**Expected behavior**
Expected sampling interval is 3.90625 ms (1000 ms / 256 Hz); actual sampling interval is 4.0 ms (1000 ms / 250 Hz).

Andrey1994, 2021-09-05:
Timestamps for neurosity are sent from device, we dont do anything with them. If its a problem it is inside neurosity streamer app

closed 2021-09-05 by Andrey1994

tstephansen, 2022-01-14:
@sjburwell Did you ever get this figured out? I’m also trying to use the crown with brainflow and when I dumped the data to a CSV I’m getting 507-514 samples per second on each channel. The sample rate is showing as 256 in the Neurosity app as well as from the BoardShim. Been pulling my hair out all day trying to figure this out.

sjburwell, 2022-01-16:
Hi @tstephansen - Unfortunately, I have not found a solution within eeg-notebooks. I think part of the issue arises because timestamps from Neurosity's device are millisecond-precision integer values (last time I checked), which was fine for the 250 sample-rate of DK1/Notion2 (sampling interval 4 ms), but not for a 256 sample-rate Crown (sampling interval 3.90625 ms).

Andrey1994, 2022-01-16:
@alexcastillo should we change sampling rate for crown from 256 to 512?

Andrey1994, 2022-01-16:
feel free to send a PR with fix https://github.com/brainflow-dev/brainflow/blob/master/src/board_controller/brainflow_boards.cpp#L340

tstephansen, 2022-01-17:
After looking into it further I'm not actually getting that many samples per second. It looks like I was actually just getting the same data twice. I'm going to need to dig further into this but I think the problem must be on my end and not with brainflow.