brainflow-dev/brainflow #348: Fix all security warnings

Andrey1994, 2021-09-22:
Need to fix all security alerts from https://lgtm.com/projects/g/brainflow-dev/brainflow/?mode=list

khoinguyen-hub, 2021-09-23:
Hi, I would like to take on this issue. This is my first time doing open-source. Will all I need to do is to resolve all of the alers? Any guidance would be appreciated!

Andrey1994, 2021-09-23:
Hi @khoinguyen-hub, sure it would be great if you help with it.

<<Will all I need to do is to resolve all of the alerts?
Yes, in this particular issue there is no need to do anything else

Take a look at https://brainflow.readthedocs.io/en/stable/BrainFlowDev.html# for getting started.
You should fork this project and send a PR, after that make sure that all tests are passed

khoinguyen-hub, 2021-09-23:
Hi @Andrey1994, thank you for your immediate response and guidance.

So I forked and created a branch to work on this issue. After I'm done fixing the alerts, do I just commits and request PR? I don't understand how my code will be tested.

Andrey1994, 2021-09-23:
You should commit your changes to a branch in your fork and create a PR on GitHub(to the master branch of https://github.com/brainflow-dev/brainflow)

There are CI pipelines that will be triggered automatically for your PR. Here is how it will look like: https://github.com/brainflow-dev/brainflow/pull/345 you can expand them by clicking "show all checks" or by navigating to Actions tab

Andrey1994, 2021-09-23:
You can also run some tests manually like checking that it compiles fine https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#compilation-of-core-module-and-c-binding 

Checking for alerts what you see here https://lgtm.com/projects/g/brainflow-dev/brainflow/?mode=list cannot be done manually(at least I have no idea how to do it) and you need to send a PR

khoinguyen-hub, 2021-09-23:
Ok, thank you for your help and your time! I really appreciate it. I will do my best in fixing these alerts.

closed 2021-12-20 by Andrey1994