brainflow-dev/brainflow #354: Jetson Nano: OSError

madhavgovindgmg, 2021-09-28:
Hey!
I ran into a problem: On using BrainFlow on Jetson Nano Device it comes up with this error.

**OSError: /home/bme/.local/lib/python3.6/site-packages/brainflow/lib/libBoardController.so: cannot open shared object file: No such file or directory**

I am running the script from the [code samples](https://brainflow.readthedocs.io/en/stable/Examples.html#python-get-data-from-a-board).

[image]

Andrey1994, 2021-09-28:
Hi, make sure that you've compiled C++ code before installation of python package. https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#linux

After that run `sudo python -m pip install -U .` from python-package folder

Andrey1994, 2021-09-28:
Also, if you have ANT Neuro board it will not work on Jetson Nano, because for ANT Neuro we need to use SDK provided by ANT Neuro company as a precompiled library(we have no source code) and they haven't built it for ARM(only for x86)

closed 2021-09-28 by Andrey1994