brainflow-dev/brainflow #382: Documentation's Python installation instructions not accurate/complete (at least for Raspberry Pi)

hack-r, 2021-11-04:
**Describe the bug**
Documentation's installation instructions not accurate for Raspberry Pi

**To Reproduce**
Try to install and use brainflow on Pi

**Info (please complete the following information):**

[According to the docs,](https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#python) to install brainflow for Python all you should need to do is 

`python -m pip install brainflow`

or

```
cd python-package
python -m pip install -U .
```

Although not mentioned in the docs (at least not in the linked docs), the pip version won't work on Pi so you're supposed to build it in the `python-package` folder (e.g. option 2 above). This appears to succeed, i.e. 

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/Documents/brainflow/python-package
Requirement already satisfied, skipping upgrade: nptyping in /usr/local/lib/python3.7/dist-packages (from brainflow==0.0.1) (1.4.4)
Requirement already satisfied, skipping upgrade: numpy in /usr/lib/python3/dist-packages (from brainflow==0.0.1) (1.16.2)
Requirement already satisfied, skipping upgrade: typish>=1.7.0 in /usr/local/lib/python3.7/dist-packages (from nptyping->brainflow==0.0.1) (1.9.3)
Building wheels for collected packages: brainflow
  Running setup.py bdist_wheel for brainflow ... done
  Stored in directory: /home/pi/.cache/pip/wheels/5f/71/6f/131148a08a5e49a4a3a6b69d0d1583f70019b6805c953c7a6d
Successfully built brainflow
Installing collected packages: brainflow
Successfully installed brainflow-0.0.1
```
However, it actually is not successful:

`sudo python3 ../tests/python/brainflow_get_data.py`
```
Traceback (most recent call last):
  File "../tests/python/brainflow_get_data.py", line 57, in <module>
    main()
  File "../tests/python/brainflow_get_data.py", line 11, in main
    BoardShim.enable_dev_board_logger()
  File "/usr/local/lib/python3.7/dist-packages/brainflow/board_shim.py", line 489, in enable_dev_board_logger
    cls.set_log_level(LogLevels.LEVEL_TRACE.value)
  File "/usr/local/lib/python3.7/dist-packages/brainflow/board_shim.py", line 472, in set_log_level
    res = BoardControllerDLL.get_instance().set_log_level_board_controller(log_level)
  File "/usr/local/lib/python3.7/dist-packages/brainflow/board_shim.py", line 140, in get_instance
    cls.__instance = cls()
  File "/usr/local/lib/python3.7/dist-packages/brainflow/board_shim.py", line 166, in __init__
    'Dynamic library %s is missed, did you forget to compile brainflow before installation of python package?' % full_path)
FileNotFoundError: Dynamic library /usr/local/lib/python3.7/dist-packages/brainflow/lib/libBoardController.so is missed, did you forget to compile brainflow before installation of python package?
```

At this point you might find [these instructions](https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#compilation-of-core-module-and-c-binding), although there's nothing in the docs that makes it clear you need to do this. So then you run `python3 -m pip install cmake`, which succeeds 

```
python3 -m pip install cmake
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cmake
  Downloading https://www.piwheels.org/simple/cmake/cmake-3.21.4-cp37-cp37m-linux_armv7l.whl (14.6MB)
    100% |████████████████████████████████| 14.6MB 33kB/s
Installing collected packages: cmake
Successfully installed cmake-3.21.4
```

but, even if you take care to add the cmake folder to your path, after you `cd tools` the next command, `python3 build.py` still fails, with or without `sudo`:

```
You need to install CMake first. Run: python -m pip install cmake
Traceback (most recent call last):
  File "build.py", line 225, in <module>
    main()
  File "build.py", line 218, in main
    check_deps()
  File "build.py", line 90, in check_deps
    raise e
  File "build.py", line 87, in check_deps
    subprocess.check_output(['cmake', '--version'])
  File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.7/subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'cmake': 'cmake'
```

You can double check that cmake is installed:

```
python3 -m pip install cmake
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: cmake in /home/pi/.local/lib/python3.7/site-packages (3.21.4)

```
I had this same sequence of errors on multiple Pi's. Although I don't recall how, I eventually found a solution before. I'd appreciate any help in recovering that solution, but more importantly I think the community would really benefit a lot if the docs were updated to reflect whatever series of commands is needed to make this work.

Andrey1994, 2021-11-05:
if due to some reason cmake installation from pip doesn't work you can install cmake from apt-get or other package managers or from cmake official website https://cmake.org/

Andrey1994, 2021-11-05:
For raspberry I think apt-get is the best option

Andrey1994, 2021-11-05:
And we have info about compilation on raspberry here https://brainflow.org/get_started/?platform=devices&language=python&iot=raspberry-pi&

hack-r, 2021-11-05:
Thank you, Andrew. It just links to the same docs I quoted though, so it won’t work if followed directly right? At the very least we should specify that it needs the apt version of cmake instead of pip (I will verify that solution later this morning).

hack-r, 2021-11-05:
I'm having the same issue after re-installing with cmake. This is currently the 2nd to most recent Slack thread on #ask-help if it's easier to troubleshoot that way.

hack-r, 2021-11-05:
We got it working! This is my suggested small change to the Docs.

I would change this:

> # Installation Instructions
> ## Python
> 
> Please, make sure to use Python 3+. Next, install the latest release from PYPI with the following command in terminal
> 
> `python -m pip install brainflow`
> 
> If you want to install it from source files or build unreleased version from Github, you should compile core module first and run
> 
> ```
> cd python-package
> python -m pip install -U .
> ```
> 

to something like this

> # Installation Instructions
> ## Python
> 
> Please, make sure to use Python 3+. Next, install the latest release from PYPI with the following command in terminal
> 
> `python -m pip install brainflow`
> 
> If you want to install it from source files or build unreleased version from Github, you should compile core module first and run
> 
> ```
> cd python-package
> python -m pip install -U .
> ```
> 
> ### Raspberry Pi
>
> Raspberry Pi users will need to first install the **apt** version of cmake, i.e. `sudo apt install cmake`.
>
> Next, compile the package from source: `sudo python3 ../tools/build.py`
>
> Install the Python package within the `python-package` folder:
> ```
> cd python-package
> python -m pip install -U .
> ```

Andrey1994, 2021-11-05:
https://github.com/brainflow-dev/brainflow/commit/acf37b25e0595691f258b48dc77ab2419f58472d

closed 2021-11-05 by Andrey1994