brainflow-dev/brainflow #398: Check that Muse works after firmware update

Andrey1994, 2021-12-06:
Maybe `write_command` should be used instead `write_request`. If so, we need to get firmware version in `prepare_session` method

Andrey1994, 2021-12-08:
just realized, we cannot get fw version because we need to send a command for it and if the way we send commands to device depends on firmware we cannot do it programmatically

Andrey1994, 2021-12-08:
Tested with 1.0.6 version and it works wo any changes(tested on macos 10.15 and windows 11), if more people complain about ubuntu maybe will add ubuntu specific handling

closed 2021-12-08 by Andrey1994

Zibri, 2022-06-21:
off topic: does anybody know if there is a repository somewhere for MUSE firmwares?

Andrey1994, 2022-06-21:
I believe its private

Zibri, 2022-06-22:
> I believe its private

probably, but the MUSE 2 is "obsolete" and perhaps someone saved it.