brainflow-dev/brainflow #425: A detailed tutorial of windows configuration environment is suggested

aimzs, 2022-03-18:
I want to know the detailed tutorial of using cmake in Windows configuration for C + + binding；
Including the detailed configuration above.

Is Cmake directly under the root directory？
I tried for a long time without success

aimzs, 2022-03-18:
The main reason is that brianflow cannot be found in cmakelist in test
[image]
find_package (
    brainflow CONFIG REQUIRED
)

Andrey1994, 2022-03-18:
Hi,

https://brainflow.readthedocs.io/en/stable/BuildBrainFlow.html#compilation-of-core-module-and-c-binding

It prints exact cmake commands, you can run it and check the output

Andrey1994, 2022-03-18:
Or check the source code of build.py

closed 2022-03-18 by Andrey1994

aimzs, 2022-03-18:
I already checked  the source code of build.py.  
i have a problem for cpp test.

Andrey1994, 2022-03-18:
Check this https://brainflow.readthedocs.io/en/stable/Examples.html#id1

Andrey1994, 2022-03-18:
For windows its almost the same

Andrey1994, 2022-03-18:
https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/run_windows.yml#L95 example from ci

aimzs, 2022-03-18:
ok,i want to try. thanks