brainflow-dev/brainflow #433: BrainFlow operations giving different results than MNE and SciPy

shreyasnisal, 2022-05-02:
I'm working with EEG data and wanted to compare the results from MNE and BrainFlow. I wrote the same operations using the two libraries, and am getting different results. The below code is an example of a detrend operation, with import statements and a call to main omitted for brevity.

```
def main():
	data = np.loadtxt('data.txt')

	# brainflow_detrend(data)
	# mne_detrend(data)

def brainflow_detrend(data):
	for channel in data:
		DataFilter.detrend(channel, DetrendOperations.LINEAR.value)

	plt.plot(data)
	plt.show()	

def mne_detrend(data):
	sfreq = 256
	electrodes = ['F3', 'F4', 'Fp1', 'Fp2', 'F7', 'F8', 'AF3', 'AF4']
	info = mne.create_info(electrodes, sfreq, 'eeg')
	evoked = mne.EvokedArray(data, info)
	evoked.detrend(1)

	plt.plot(evoked.data)
	plt.show()
```

I'm using data without using the BrainFlow built-in functions to get data from boards, and one of the first issues is that the detrend function gives an error with NDArray and insists on a 1D Array. This does not seem to match the documentation, a screenshot of which is below.

[image]

The use of detrend in BrainFlow and MNE gives the following two plots

**BrainFlow**
[image]

**MNE**
[image]

I also tried out the detrending using SciPy, and it gave me the same results as MNE.
This looks like BrainFlow is performing the detrend operation in a completely different way. Any help would be appreciated!

Andrey1994, 2022-05-02:
Hi,

its not a surprise that mne returns the same as scipy because mne just calls method from scipy https://github.com/mne-tools/mne-python/blob/62143cbeb12a822e08833941c4e0f1d450d35240/mne/filter.py#L1616 

<<gives an error with NDArray and insists on a 1D Array. This does not seem to match the documentation, a screenshot of which is below.

There is no nptyping type for 1d array, so its written as NDArray in docs, I dont see the issue here

Andrey1994, 2022-05-02:
Implementation inside brainflow is simple https://github.com/brainflow-dev/brainflow/blob/master/src/data_handler/data_handler.cpp#L1067 and I dont see any bugs here

Andrey1994, 2022-05-02:
I also created a test for it and everything looks fine

```python
import argparse
import time
import brainflow
import numpy as np

import pandas as pd
import matplotlib

matplotlib.use('Agg')
import matplotlib.pyplot as plt

from brainflow.board_shim import BoardShim, BrainFlowInputParams, LogLevels, BoardIds
from brainflow.data_filter import DataFilter, FilterTypes, AggOperations, NoiseTypes, DetrendOperations

def main():
    BoardShim.enable_dev_board_logger()

    # use synthetic board for demo
    params = BrainFlowInputParams()
    board_id = BoardIds.SYNTHETIC_BOARD.value
    board = BoardShim(board_id, params)
    board.prepare_session()
    board.start_stream()
    BoardShim.log_message(LogLevels.LEVEL_INFO.value, 'start sleeping in the main thread')
    time.sleep(5)
    data = board.get_current_board_data(256)
    board.stop_stream()
    board.release_session()

    eeg_channels = BoardShim.get_eeg_channels(board_id)

    # add linear trend to data
    for count, channel in enumerate(eeg_channels):
        for i in range(len(data[channel])):
            data[channel][i] = data[channel][i] + i * count 

    df = pd.DataFrame(np.transpose(data))
    plt.figure()
    df[eeg_channels].plot(subplots=True)
    plt.savefig('before_processing.png')

    for channel in eeg_channels:
        DataFilter.detrend(data[channel], DetrendOperations.LINEAR.value)

    df = pd.DataFrame(np.transpose(data))
    plt.figure()
    df[eeg_channels].plot(subplots=True)
    plt.savefig('after_processing.png')

if __name__ == "__main__":
    main()
```

Before:
[image]

After:
[image]

closed 2022-05-02 by Andrey1994

shreyasnisal, 2022-05-02:
Thank you for your prompt reply!

Since there are no issues in the detrending, is there any specific reason why SciPy and BrainFlow are giving different results based on the code used earlier?

Andrey1994, 2022-05-02:
I dont fully understand how you tested it... you can take a look at scipy code and at brainflow code to compare. I've checked briefly there are a few more options but I didnt notice anything specific at the first sight