brainflow-dev/brainflow #434: "terminate called without an active exception" with Muse 2 and Ubuntu

barlac, 2022-05-07:
**Describe the bug**
Got `terminate called without an active exception`, and subsequent crash when connecting to muse 2 on ubuntu.
```

[2022-05-07 22:54:49.125] [board_logger] [info] incoming json: {
    "file": "",
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "00:55:DA:B7:86:B5",
    "other_info": "",
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2022-05-07 22:54:49.126] [board_logger] [trace] Board object created 38
[2022-05-07 22:54:49.126] [board_logger] [info] Use timeout for discovery: 5
[2022-05-07 22:54:49.126] [board_logger] [debug] use dyn lib: /home/username/projects/realtime_EEG/muse_venv/lib/python3.8/site-packages/brainflow/lib/libsimpleble-c.so
[2022-05-07 22:54:49.130] [board_logger] [info] found 2 BLE adapter(s)
[2022-05-07 22:54:49.733] [board_logger] [trace] address C8:CD:41:44:81:5C
[2022-05-07 22:54:49.733] [board_logger] [trace] identifier LE_WH-1000XM4
[2022-05-07 22:54:50.175] [board_logger] [trace] address 51:8D:0C:19:B4:E9
[2022-05-07 22:54:50.175] [board_logger] [trace] identifier 
[2022-05-07 22:54:50.644] [board_logger] [trace] address 00:55:DA:B7:86:B5
[2022-05-07 22:54:50.644] [board_logger] [trace] identifier Muse-86B5
[2022-05-07 22:54:50.644] [board_logger] [info] Found Muse device
[2022-05-07 22:54:51.520] [board_logger] [info] Connected to Muse Device
[2022-05-07 22:54:52.522] [board_logger] [trace] found servce 00001801-0000-1000-8000-00805f9b34fb
[2022-05-07 22:54:52.522] [board_logger] [trace] found characteristic 00002a05-0000-1000-8000-00805f9b34fb
[2022-05-07 22:54:52.523] [board_logger] [trace] found servce 0000fe8d-0000-1000-8000-00805f9b34fb
[2022-05-07 22:54:52.523] [board_logger] [trace] found characteristic 273e0001-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.523] [board_logger] [info] found control characteristic
[2022-05-07 22:54:52.523] [board_logger] [trace] found characteristic 273e0008-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.523] [board_logger] [trace] found characteristic 273e0009-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.524] [board_logger] [trace] found characteristic 273e000a-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.525] [board_logger] [trace] found characteristic 273e000b-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.525] [board_logger] [trace] found characteristic 273e0002-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.525] [board_logger] [trace] found characteristic 273e0003-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.525] [board_logger] [trace] found characteristic 273e0004-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.526] [board_logger] [trace] found characteristic 273e0005-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.526] [board_logger] [trace] found characteristic 273e0006-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.527] [board_logger] [trace] found characteristic 273e0007-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.527] [board_logger] [trace] found characteristic 273e000c-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.527] [board_logger] [trace] found characteristic 273e000d-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.527] [board_logger] [trace] found characteristic 273e000e-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.527] [board_logger] [trace] found characteristic 273e000f-4c4d-454d-96be-f03bac821358
[2022-05-07 22:54:52.528] [board_logger] [trace] found characteristic 273e0010-4c4d-454d-96be-f03bac821358
terminate called without an active exception
```

This is gbd output when running `get_data_demo` cpp test with a board_id of 38.

```
(muse_venv) username@auc215223:~/projects/realtime_EEG/brainflow/tests/cpp/config_board$ sudo gdb --args ./config_board --board-id 38
(gdb) r
Starting program: /home/username/projects/realtime_EEG/brainflow/tests/cpp/config_board/config_board --board-id 38
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[2022-05-07 22:58:19.824] [board_logger] [info] incoming json: {"file":"","ip_address":"","ip_port":0,"ip_protocol":0,"mac_address":"","other_info":"","serial_number":"","serial_port":"","timeout":0}
[2022-05-07 22:58:19.824] [board_logger] [trace] Board object created 38
[2022-05-07 22:58:19.824] [board_logger] [info] Use timeout for discovery: 5
[2022-05-07 22:58:19.824] [board_logger] [debug] use dyn lib: /home/username/projects/realtime_EEG/brainflow/installed/lib/libsimpleble-c.so
[New Thread 0x7ffff73b0700 (LWP 118813)]
[2022-05-07 22:58:19.964] [board_logger] [info] found 2 BLE adapter(s)
[2022-05-07 22:58:20.806] [board_logger] [trace] address C8:CD:41:44:81:5C
[2022-05-07 22:58:20.806] [board_logger] [trace] identifier LE_WH-1000XM4
[2022-05-07 22:58:20.910] [board_logger] [trace] address 5F:96:D0:8C:AD:93
[2022-05-07 22:58:20.910] [board_logger] [trace] identifier 
[2022-05-07 22:58:21.010] [board_logger] [trace] address 51:8D:0C:19:B4:E9
[2022-05-07 22:58:21.010] [board_logger] [trace] identifier 
[... 36 lines cut ...]
    at /home/username/projects/realtime_EEG/brainflow/third_party/SimpleBLE/src/PeripheralSafe.cpp:86
#6  0x00007ffff7676063 in simpleble_peripheral_write_request (handle=0x7ffff00174b0, service=..., characteristic=..., data=0x7fffffffd8a0 "\003v1\nUU", data_length=4)
    at /home/username/projects/realtime_EEG/brainflow/third_party/SimpleBLE/src_c/peripheral.cpp:213
#7  0x00007ffff7e9e3e6 in BLELibBoard::simpleble_peripheral_write_request (this=0x5555555d8a70, handle=0x7ffff00174b0, service=..., characteristic=..., data=0x7fffffffd8a0 "\003v1\nUU", data_length=4)
    at /home/username/projects/realtime_EEG/brainflow/src/board_controller/ble_lib_board.cpp:436
#8  0x00007ffff7ea2f14 in Muse::config_board (this=0x5555555d8a70, config="v1") at /home/username/projects/realtime_EEG/brainflow/src/board_controller/muse/muse.cpp:499
#9  0x00007ffff7ea25f4 in Muse::prepare_session (this=0x5555555d8a70) at /home/username/projects/realtime_EEG/brainflow/src/board_controller/muse/muse.cpp:378
#10 0x00007ffff7dfaa8c in prepare_session (board_id=38, 
    json_brainflow_input_params=0x5555555d86f0 "{\"file\":\"\",\"ip_address\":\"\",\"ip_port\":0,\"ip_protocol\":0,\"mac_address\":\"\",\"other_info\":\"\",\"serial_number\":\"\",\"serial_port\":\"\",\"timeout\":0}") at /home/username/projects/realtime_EEG/brainflow/src/board_controller/board_controller.cpp:248
#11 0x000055555556a0c6 in BoardShim::prepare_session (this=0x5555555d8350) at /home/username/projects/realtime_EEG/brainflow/cpp-package/src/board_shim.cpp:106
#12 0x0000555555567327 in main (argc=3, argv=0x7fffffffe558) at /home/username/projects/realtime_EEG/brainflow/tests/cpp/config_board/src/config_board.cpp:34
```

**To Reproduce**
Call the `get_data_demo` cpp test using Ubuntu Linux with a board_id of 38 using a Muse 2.

**Info (please complete the following information):**
 - Board Id: **38**
 - OS running on your PC: **Ubuntu Linux Focal**
 - Programming Language and its version **Python 3**
 - Architecture(x86, x64, ARM, etc) **x86_64**
 - **BrainFlow installed from source with commit ecb7fb4** 

**Additional context**
This is similar to #407, however, the steps were followed to install BrainFlow from source and a similar error was observed. The output from gbd is shown. It appears that the error occurs (i.e., while attempting connection) when the "simpleble_peripheral_write_command" is called from the BLELibBoard::dll_loader, more detailed info shown above in the gbd output.

Andrey1994, 2022-05-08:
Hi and thanks a lot for gdb backtrace!

I have no ubuntu to test... in fact for linux I can test it only using raspberry pi.
It looks like an issue in SimpleBLE but I dont fully understand the backtrace, line 86 is:
```
bool SimpleBLE::Safe::Peripheral::write_request(BluetoothUUID service, BluetoothUUID characteristic,
                                                ByteArray data) noexcept {
    try {
        SimpleBLE::Peripheral::write_request(service, characteristic, data);
        return true;
    } catch (const SimpleBLE::Exception::BaseException&) {
        return false;
    }
}
```
Can it be that there is an error in another thread? Could you switch to other threads if any and check backtrace there too? If I remember correctly gdb commands for it are `info threads` and `thread %num%`

barlac, 2022-05-11:
I had a look but wasn't able to work it out. I've attached the bt for the other thread.

```
(gdb) info threads
  Id   Target Id                                         Frame 
* 1    Thread 0x7ffff7708740 (LWP 199535) "config_board" __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
  2    Thread 0x7ffff73b0700 (LWP 199539) "config_board" 0x00007ffff796226f in __GI___clock_nanosleep (clock_id=clock_id@entry=0, 
    flags=flags@entry=0, req=0x7ffff73afdb0, rem=0x7ffff73afdb0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
(gdb) thread 2
[Switching to thread 2 (Thread 0x7ffff73b0700 (LWP 199539))]
#0  0x00007ffff796226f in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7ffff73afdb0, rem=0x7ffff73afdb0)
    at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
78	../sysdeps/unix/sysv/linux/clock_nanosleep.c: No such file or directory.
(gdb) bt
#0  0x00007ffff796226f in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7ffff73afdb0, rem=0x7ffff73afdb0)
    at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
#1  0x00007ffff7967ef7 in __GI___nanosleep (requested_time=<optimized out>, remaining=<optimized out>) at nanosleep.c:27
#2  0x00007ffff768fc1b in std::this_thread::sleep_for<long, std::ratio<1l, 1000000l> > (__rtime=...) at /usr/include/c++/9/thread:378
#3  0x00007ffff768f6c3 in SimpleBLE::Bluez::async_thread_function (this=0x7ffff7707600 <SimpleBLE::Bluez::get()::instance>)
    at /home/lbarnes/projects/realtime_EEG/brainflow/third_party/SimpleBLE/src/linux/Bluez.cpp:31
#4  0x00007ffff76906fe in std::__invoke_impl<void, void (SimpleBLE::Bluez::*)(), SimpleBLE::Bluez*> (
    __f=@0x555555604390: (void (SimpleBLE::Bluez::*)(class SimpleBLE::Bluez * const)) 0x7ffff768f65c <SimpleBLE::Bluez::async_thread_function()>, 
    __t=@0x555555604388: 0x7ffff7707600 <SimpleBLE::Bluez::get()::instance>) at /usr/include/c++/9/bits/invoke.h:73
#5  0x00007ffff769062c in std::__invoke<void (SimpleBLE::Bluez::*)(), SimpleBLE::Bluez*> (
    __fn=@0x555555604390: (void (SimpleBLE::Bluez::*)(class SimpleBLE::Bluez * const)) 0x7ffff768f65c <SimpleBLE::Bluez::async_thread_function()>)
    at /usr/include/c++/9/bits/invoke.h:95
#6  0x00007ffff769058b in std::thread::_Invoker<std::tuple<void (SimpleBLE::Bluez::*)(), SimpleBLE::Bluez*> >::_M_invoke<0ul, 1ul> (
    this=0x555555604388) at /usr/include/c++/9/thread:244
#7  0x00007ffff7690542 in std::thread::_Invoker<std::tuple<void (SimpleBLE::Bluez::*)(), SimpleBLE::Bluez*> >::operator() (this=0x555555604388)
    at /usr/include/c++/9/thread:251
#8  0x00007ffff7690522 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (SimpleBLE::Bluez::*)(), SimpleBLE::Bluez*> > >::_M_run (
    this=0x555555604380) at /usr/include/c++/9/thread:195
#9  0x00007ffff7b68de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff786a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#11 0x00007ffff79a4163 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

Andrey1994, 2022-05-11:
I've asked SimpleBLE developers maybe they have some insights.

I cannot test locally because of https://github.com/openwrt/mt76/issues/548#issuecomment-881869386 bluetooth just doesnt work for me, I will give it another try with the latest kernel.

Another thing you can try in the meantime - build SimpleBLE from upstream using latest master and copypaste resulting .so file instead .so compiled by brainflow(in brainflow as of right now older version is used). If it helps then we will just need to update SimpleBLE fork in brainflow

Andrey1994, 2022-05-16:
I've updated simpleble in a linked PR, there were plenty of changes especially for linux. Could you test it and let me know if the issue is solved or not?

closed 2022-05-16 by Andrey1994