brainflow-dev/brainflow #515: How do I enable PPG on my Muse 2 Headband

ChilloutCharles, 2022-06-27:
I am able to get the channels for PPG but they're all returning zero. What step am I missing?

Andrey1994, 2022-06-27:
need to change default preset for muse board https://github.com/brainflow-dev/brainflow/blob/master/python_package/examples/tests/muse_ppg.py

ChilloutCharles, 2022-06-27:
Would there be any documentation on using the config line? I am also getting the same issue when trying to pull up the resist channels for my brainbit headband

Andrey1994, 2022-06-27:
https://github.com/brainflow-dev/brainflow/blob/master/python_package/examples/tests/brainbit_resist.py here is an example for brainbit.

Arguments for `config_board` method are device specific and depend on hardware, so it makes it harder to document it. Feel free to add some sentences to SupportedBoards page and send a PR, it will be good