brainflow-dev/brainflow #516: add an option to specify multiple streamers per board

Andrey1994, 2022-06-29:
it should be possible to write smth like:

```
board->add_streamer("file://data.csv:w");
board->add_streamer("file://data2.csv:w");
board->add_streamer("streaming_board://225.1.1.1:6677");
```

Inside board class currently these is `std::map<int, Streamer *> streamers;` need to convert it to smth like `std::map<int, std::vector<Streamer *>> streamers;` `int` here represents preset

Wameuh, 2022-09-13:
Hi! I am trying to work on this issue, but I have a question.

About the method Board::push_package, do you want that a specific streamer should be added as an argument, or every stream data should be printed?

I think, at some point some documentation to run tests locally should be great for new contributors.

Andrey1994, 2022-09-13:
Hi and thanks for working on it!

Could you clarify what you mean by "do you want that a specific streamer should be added as an argument, or every stream data should be printed?"?

Andrey1994, 2022-09-13:
I think `push_package(double *package, int preset)` should work like this:

- push package to the correct data buffer for specified preset
- iterate over all registered streamers for this preset and call `stream_data` for each of them

Streamers can be registered by calling `add_streamer` from the user code. Also in `start_stream` method there is a legacy way to add streamer for default preset, it remains there only for backward compatibility.

About tests, for this task you will need to write one by yourself. We definitely will need c++ tests and it should be executed with valgrind and simple python test for demo. Instructions how to run them can be found in Code Samples page at BrainFlow docs

To add test case to CI pipelines you will need to add it here https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/valgrind.yml for valgrind, python test can be added to https://github.com/brainflow-dev/brainflow/blob/master/.github/workflows/run_unix.yml There is no need to add tests for other languages because its internal change which doesnt affect bindings

Wameuh, 2022-09-13:
Thanks for the information. You have responded to my question about push_package and it was the way I thought. Thanks!

About tests. Do I have to create them or not? Because even if I am familiar with GTest, I am not with CI pipelines. But I can learn, it will just need some time to be able to create them.

Wameuh, 2022-09-13:
I have creating a PR: https://github.com/brainflow-dev/brainflow/pull/550

closed 2022-11-03 by Andrey1994