brainflow-dev/brainflow #522: Muse 2 disconnection issue in Python

ChilloutCharles, 2022-07-15:
**Describe the bug**
Muse 2 Headband stops sending data after a few minutes with no error

**To Reproduce**
using a modified version of the brainflow_get_data test attached: [brainflow_get_data.py.txt](https://github.com/brainflow-dev/brainflow/files/9122887/brainflow_get_data.py.txt)

**Info (please complete the following information):**
 - Board Id : 38
 - OS running on your PC: Windows 10 Version 21H2
 - Programming Language and its version: Python 3.8.5
 - Architecture(x86, x64, ARM, etc): x64
 - Python Pip List: [piplist.txt](https://github.com/brainflow-dev/brainflow/files/9122872/piplist.txt)

**Expected behavior**
Muse 2 headband should continue to send information.

**Additional context**
Logfile attached: [board_log_file.txt](https://github.com/brainflow-dev/brainflow/files/9122860/board_log_file.txt)

Andrey1994, 2022-07-15:
there will be a release next week with plenty of fixes for muse boards, would be good if you test it one more time after it

ChilloutCharles, 2022-07-15:
> there will be a release next week with plenty of fixes for muse boards, would be good if you test it one more time after it

Thanks! I'll be looking forward to it. Will close when I encounter no issues with it

Andrey1994, 2022-07-16:
https://github.com/brainflow-dev/brainflow/issues/393 it also can be relevant

ChilloutCharles, 2022-07-25:
I use my muse with a dongle, so I should be good to avoid that. Any news on that release?

Andrey1994, 2022-07-25:
will be published today, I already merged this pr to master

Andrey1994, 2022-07-25:
tried with latest release and muse s. cannot reproduce, closing this now. feel free to reopen if its still an issue also check the battery first

closed 2022-07-25 by Andrey1994

ChilloutCharles, 2022-07-26:
Looks good so far, but somehow I lost ppg with my muse 2. Will write a different issue later today.

Andrey1994, 2022-07-26:
Its not an issue, read release post