brainflow-dev/brainflow #525: Neurosity crown streaming error

JohnGriffiths, 2022-07-21:
**Describe the bug**

Brainflow streaming with neurosity crown errors for unknown reasons.

**To Reproduce**
To report a bug you need to create a simple example that uses only BrainFlow API or use one of the tests we already have.  In this example, you need to call `enable_dev_board_logger ()` and share the output with us. Also, make sure the latest version is used. 

See screenshow below for code + outputs
Code lines to run to reproduce = 

```python
from brainflow.board_shim import BoardShim,BoardIds,BrainFlowInputParams
params = BrainFlowInputParams()
shimmer = BoardShim(input_params=params,board_id=BoardIds.CROWN_BOARD)
shimmer.enable_dev_board_logger()
shimmer.prepare_session()
shimmer.start_stream()
```

**Info (please complete the following information):**
 - Board Id    -  **CROWN_BOARD**
 - OS running on your PC - **Windows**
 - Programming Language and its version - **Python 3.7**
 - Architecture(x86, x64, ARM, etc) - **x64**

**Expected behavior**
A clear and concise description of what you expected to happen.

`start_stream()` does not end with error. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Additional context**
Add any other context about the problem here.

- Seen across >3 computers now, with latest brainflow version
- Persists after turning off firewall
- Isn't fixed by adding mac address or IP address in params

---

Thanks!

Andrey1994, 2022-07-21:
@alexcastillo could you take a look? I have no neurosity boards and cannot test. 10060 is a socket timeour error, so no data was sent from device

Andrey1994, 2022-07-25:
maybe it makes sense for you to ping neurosity guys directly about this issue...

leejr72, 2022-08-03:
Receiving similar results off macOS.

First run is when the crown is in sleep mode and only sending the device id info:

admin@forest brainflow % python3 brainflow-test.py
[2022-08-03 14:41:25.834] [board_logger] [info] incoming json: {
    "board_id": 23,
    "file": "",
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": -100,
    "other_info": "",
    "preset": 0,
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2022-08-03 14:41:25.834] [board_logger] [trace] Board object created 13
[2022-08-03 14:41:25.834] [board_logger] [debug] Use IP port 9000
[2022-08-03 14:41:26.208] [board_logger] [info] received first package with 188 bytes streaming is started
[2022-08-03 14:41:26.209] [board_logger] [debug] start streaming
**[2022-08-03 14:41:26.209] [board_logger] [trace] Unknown msg: /neurosity/notion/42e9fd46d4d05ac597588c974f17a8ae/info
[2022-08-03 14:41:27.130] [board_logger] [trace] Unknown msg: /neurosity/notion/42e9fd46d4d05ac597588c974f17a8ae/info
[]**
admin@forest brainflow % 

Wireshark packet capture:
Open Sound Control Encoding
    Message: /neurosity/notion/42e9fd46d4d05ac597588c974f17a8ae/info ,ssssssiis
        Header
        String: 42e9fd46d4d05ac597588c974f17a8ae
        String: Crown-42E
        String: Crown 3
        String: Crown
        String: 3
        String: Neurosity, Inc
        Int32: 256
        Int32: 8
        String: CP3,C3,F5,PO3,PO4,F6,C4,CP4

When crown is normal streaming 
admin@forest brainflow % python3 brainflow-test.py
[2022-08-03 14:44:54.229] [board_logger] [info] incoming json: {
    "board_id": 23,
    "file": "",
    "ip_address": "",
    "ip_port": 0,
    "ip_protocol": 0,
    "mac_address": "",
    "master_board": -100,
    "other_info": "",
    "preset": 0,
    "serial_number": "",
    "serial_port": "",
    "timeout": 0
}
[2022-08-03 14:44:54.229] [board_logger] [trace] Board object created 13
[2022-08-03 14:44:54.229] [board_logger] [debug] Use IP port 9000
[2022-08-03 14:44:54.288] [board_logger] [info] received first package with 128 bytes streaming is started
[2022-08-03 14:44:54.288] [board_logger] [debug] start streaming
[[ 0.00000000e+00  1.00000000e+00  2.00000000e+00]
 [-1.21178867e+04  7.36004932e+03  2.67274121e+04]
 [-8.58648926e+03  1.12192119e+04  3.15515664e+04]
 [-3.72377832e+03  1.61094160e+04  2.82363555e+04]
 [ 9.71784570e+03  2.82935527e+04  2.38643535e+04]
 [-6.33142236e+03  1.76974180e+04  3.92924883e+04]
 [-2.04142285e+04  2.60268188e+03  3.75929062e+04]
 [-3.56229906e+05 -3.32430125e+05 -2.81671125e+05]
 [-3.47194570e+04 -9.20775684e+03  3.91276680e+04]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 1.65955949e+12  1.65955949e+12  1.65955949e+12]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00]]
admin@forest brainflow % 

Never see the device OSC device ID packets decoded.

Cheers

leejr72, 2022-08-03:
Also noticed the Board object created is 13 even though 23 is called out.

code used for test
import brainflow

from brainflow import BoardIds
from brainflow.board_shim import BoardShim, BrainFlowInputParams
from brainflow.data_filter import DataFilter, FilterTypes, AggOperations

def main ():
    board_id = BoardIds.CROWN_BOARD.value # or BoardIds.NOTION_2_BOARD.value or BoardIds.NOTION_1_BOARD.value
    params = BrainFlowInputParams ()
    params.board_id = board_id
    BoardShim.enable_dev_board_logger ()
    board = BoardShim (board_id, params)
    board.prepare_session ()
    board.start_stream ()
    data = board.get_board_data ()
    board.stop_stream ()
    board.release_session ()

    print (data)

if __name__ == "__main__":
    main ()

Andrey1994, 2022-08-03:
https://github.com/brainflow-dev/brainflow/issues/525#issuecomment-1204470416 it doesnt look like the issue, it seems like everything works fine for you. We dont parse some info packages, only data packages and its expected, you can ignore trace log messages, they are for debugging and developers

Andrey1994, 2022-08-03:
https://github.com/brainflow-dev/brainflow/issues/525#issuecomment-1204471833 I can fix it but it also doesnt really matter since all neurosity devices stream the same data we dont make any difference in brainflow between them in fact

Andrey1994, 2022-08-03:
@JohnGriffiths I've just tested with the emulator https://github.com/neurosity/notion-osc-server and it works fine for me. Could you give it a try with emulator? If it doesnt work for you lets compare setups if it works in emulator it definitely a question for neurosity guys(and unfortunately for some reason they do not answer), there is nothing else I can do with it.

alexcastillo, 2022-08-03:
Hey everyone,

I'd love to help.

I can confirm Brainflow is working properly with the Crown.

Can you please post more information about your setup?
 - Platform (Mac/Linux/Windows)
 - OS version
 - Python version
 - Confirm that Crown OSC setting is enabled (use the Neurosity app)
 - Confirm your Crown is connected to the same Wi-Fi as your computer
 - Confirm your Crown is powered on, and not charging
 - Confirm you are not using a public network (e.g. coffee shop, co-working space, or university)

leejr72, 2022-08-03:
Understood thank you,

Andrey1994, 2022-08-09:
closing this issue, feel free to continue discussion here and find the problem with setup. but it doesnt look like the issue that can be solved by brainflow.

in theory we can replace broadcasting by plain sockets(like point to point) and maybe it will work better with some nets, but its up to @alexcastillo and deserves its own issue\task on github

closed 2022-08-09 by Andrey1994