brainflow-dev/brainflow #527: multiple bugs on data_handler::read_file and DataFilter::read_file lead to heap corruption

echel0nn, 2022-07-28:
**Describe the bug**
read_file function causes SIGSEGV, SIGBUS,SIGABRT faults when it is filled with arbitrary data and these bugs can also lead to security issues. 

**To Reproduce**
- [CPP Example](https://github.com/echel0nn/brainflow-fuzz/blob/main/harness.cpp)
- [Python Example](https://github.com/echel0nn/brainflow-fuzz/blob/main/brainoverflow.py)
- [Files](https://github.com/echel0nn/brainflow-fuzz/tree/main/crashers) 

**Info (please complete the following information):**
 - Board Id - Not applicable
 - OS running on your PC - Arch Linux x64 Linux 5.18
 - Programming Language and its version: Python 3.10.5/C++
 - Architecture(x86, x64, ARM, etc) x64
 - In case of Python full output of `pip list` command, for other languages please provide BrainFlow version used
- brainflow==5.1.0

**Expected behavior**
Must discard the given irregular data

**Screenshots**
- Given nonsense data corrupts the heap
[image]

**Additional context**
[Which-Crashes-Which Table](https://github.com/echel0nn/brainflow-fuzz/blob/main/STATS.md)
I am opening this issue before finding the root causes because I may not be available in the coming days to work on.
Note: I'm not using brainflow libraries personally, I just tested the functions how they are explained in documents. These findings can be unrelated or obsolete. Also i can provide a `Dockerfile` to fuzz `read_file` functionality continuously.
Other informations are available in this [readme](https://github.com/echel0nn/brainflow-fuzz)

Andrey1994, 2022-07-28:
hi and thanks for reporting.

Looking at the code I see one potential issue so far, somewhere here https://github.com/brainflow-dev/brainflow/blob/master/src/data_handler/data_handler.cpp#L1094 should be a check that there is the same amount of cols in each row

Andrey1994, 2022-07-28:
feel free to reopen if there is a failed test which I missed in pr

closed 2022-07-28 by Andrey1994