brainflow-dev/brainflow #531: Unable to prepare streaming session after finding Muse2

Parvfect, 2022-08-03:
**Describe the bug**
Unable to prepare the streaming session after finding Muse2 using direct calls to Brainflow. 
 

**To Reproduce**
```python
from brainflow.board_shim import BoardShim, BoardIds, BrainFlowInputParams

params = BrainFlowInputParams()
board_id = BoardIds.MUSE_2_BOARD

board = BoardShim(board_id, params)
board.enable_dev_board_logger()

board.prepare_session()
```

**Info (please complete the following information):**
 - Board Id - MUSE_2_BOARD
 - OS running on your PC - Windows 11
 - Programming Language and its version - Python 3.7.13
 - Architecture(x86, x64, ARM, etc) x64
 - In case of Python full output of `pip list` command, for other languages please provide BrainFlow version used
Package                       Version      Editable project location
----------------------------- ------------ -------------------------------------
alabaster                     0.7.12
appdirs                       1.4.4
arabic-reshaper               2.1.3
argon2-cffi                   21.3.0
argon2-cffi-bindings          21.2.0
astunparse                    1.6.3
atomicwrites                  1.4.1
attrs                         21.4.0
Babel                         2.10.3
backcall                      0.2.0
beautifulsoup4                4.11.1
bitstring                     3.1.9
bleach                        5.0.1
bleak                         0.14.3
bleak-winrt                   1.1.1
brainflow                     5.1.0
certifi                       2022.6.15
cffi                          1.15.1
charset-normalizer            2.1.0
click                         8.1.3
cmake                         3.23.3
colorama                      0.4.5
commonmark                    0.9.1
contextlib2                   21.6.0
coverage                      6.4.2
cycler                        0.11.0
debugpy                       1.6.2
decorator                     4.4.2
defusedxml                    0.7.1
docutils                      0.17.1
dukpy                         0.2.3
eeg-notebooks                 0.2          c:\users\parv\documents\eeg-notebooks
entrypoints                   0.4
enum-compat                   0.0.3
esprima                       4.0.1
et-xmlfile                    1.1.0
execnet                       1.9.0
fastjsonschema                2.16.1
filelock                      3.7.1
fonttools                     4.34.4
freetype-py                   2.3.0
future                        0.18.2
gdown                         4.5.1
gevent                        21.12.0
gitdb                         4.0.9
GitPython                     3.1.27
glfw                          2.5.4
greenlet                      1.1.2
h5py                          3.7.0
hsluv                         5.0.3
idna                          3.3
imageio                       2.20.0
imageio-ffmpeg                0.4.7
imagesize                     1.4.1
importlib-metadata            4.12.0
importlib-resources           5.9.0
iniconfig                     1.1.1
ipykernel                     6.15.1
ipython                       7.34.0
ipython-genutils              0.2.0
ipywidgets                    7.7.1
javascripthon                 0.12
jedi                          0.18.1
Jinja2                        3.0.3
joblib                        1.1.0
json-tricks                   3.15.5
jsonschema                    4.8.0
jupyter                       1.0.0
jupyter-client                7.3.4
jupyter-console               6.4.4
jupyter-core                  4.11.1
jupyterlab-pygments           0.2.2
jupyterlab-widgets            1.1.1
kiwisolver                    1.4.4
MarkupSafe                    2.1.1
matplotlib                    3.5.2
matplotlib-inline             0.1.3
mistune                       0.8.4
mne                           1.0.3
mock                          4.0.3
moviepy                       1.0.3
msgpack                       1.0.4
msgpack-numpy                 0.4.8
muselsl                       2.2.1
mypy                          0.971
mypy-extensions               0.4.3
nbclient                      0.6.6
nbconvert                     6.5.0
nbformat                      5.4.0
nbval                         0.9.6
nest-asyncio                  1.5.5
notebook                      6.4.12
nptyping                      1.4.4
numexpr                       2.8.3
numpy                         1.21.6
numpydoc                      1.4.0
openpyxl                      3.0.10
packaging                     21.3
pandas                        1.3.5
pandocfilters                 1.5.0
parso                         0.8.3
path                          16.4.0
path.py                       12.5.0
pexpect                       4.8.0
pickleshare                   0.7.5
Pillow                        9.2.0
pip                           22.1.2
pluggy                        1.0.0
pooch                         1.6.0
proglog                       0.1.10
prometheus-client             0.14.1
prompt-toolkit                3.0.30
psutil                        5.9.1
PsychoPy                      2020.2.3
psychtoolbox                  3.0.18.2
ptyprocess                    0.7.0
py                            1.11.0
pycparser                     2.21
pygatt                        4.0.5
pyglet                        1.4.10
Pygments                      2.12.0
pylsl                         1.16.0
PyOpenGL                      3.1.6
pyosf                         1.0.5
pyparsing                     3.0.9
pypiwin32                     223
pyqmix                        2021.1.2
pyriemann                     0.3
pyrsistent                    0.18.1
pyserial                      3.5
PySocks                       1.7.1
pytest                        7.1.2
pytest-cov                    3.0.0
pytest-shutil                 1.7.0
python-bidi                   0.4.2
python-dateutil               2.8.2
python-gitlab                 3.6.0
pytz                          2022.1
pywin32                       304
pyWinhook                     1.6.2
pywinpty                      2.0.6
PyYAML                        6.0
pyzmq                         23.2.0
qtconsole                     5.3.1
QtPy                          2.1.0
questplus                     2019.4
recommonmark                  0.7.1
reportlab                     3.6.11
requests                      2.28.1
requests-toolbelt             0.9.1
rst2pdf                       0.99
ruamel.yaml                   0.16.3
ruamel.yaml.clib              0.2.6
scikit-learn                  1.0.2
scipy                         1.7.3
seaborn                       0.9.0
Send2Trash                    1.8.0
setuptools                    61.2.0
six                           1.16.0
smartypants                   2.0.1
smmap                         5.0.0
snowballstemmer               2.2.0
sounddevice                   0.4.4
SoundFile                     0.10.3.post1
soupsieve                     2.3.2.post1
Sphinx                        5.1.1
sphinx-copybutton             0.5.0
sphinx-gallery                0.11.0
sphinx-rtd-theme              1.0.0
sphinx-tabs                   3.4.0
sphinxcontrib-applehelp       1.0.2
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        2.0.0
sphinxcontrib-httpdomain      1.8.0
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.5
tables                        3.7.0
termcolor                     1.1.0
terminado                     0.15.0
threadpoolctl                 3.1.0
tinycss2                      1.1.1
tomli                         2.0.1
tornado                       6.2
tqdm                          4.64.0
traitlets                     5.3.0
typed-ast                     1.5.4
types-requests                2.28.5
types-urllib3                 1.26.17
typing_extensions             4.3.0
typish                        1.9.3
urllib3                       1.26.11
versioneer                    0.22
vispy                         0.11.0
wcwidth                       0.2.5
webencodings                  0.5.1
websocket-client              1.3.3
wheel                         0.37.1
widgetsnbextension            3.6.1
wincertstore                  0.2
wxPython                      4.1.1
xarray                        0.20.2
xlrd                          2.0.1
zipp                          3.8.1
zope.event                    4.5.0
zope.interface                5.4.0

**Expected behavior**
Streaming Session is not able to be prepared due to inability to connect to device, client errors out.

**Screenshots**
Screenshot
[image]

**Additional context**
Originated when trying to run eeg-notebooks experiment N170 using the Brainflow backend.

Parvfect, 2022-08-03:
https://github.com/NeuroTechX/eeg-notebooks/issues/199

Issue opened in EEG-Notebooks

Andrey1994, 2022-08-03:
https://github.com/brainflow-dev/brainflow/issues/393 check this

JohnGriffiths, 2022-08-03:
@Andrey1994 are you saying the fix is to update to python 3.10 and it won't (ever?) work for 3.8+?

That would likely cause major dependencies headaches with other libs for us.

Andrey1994, 2022-08-03:
using windows 10 it worked for me just fine using python 3.6 and python 3.9
when I updated to windows 11 it stopped working and I spent days debugging why... here is the full story https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/31  and the only solution I've found is to update python to 3.10. I even have no idea what was changes between these python versions

Andrey1994, 2022-08-03:
Could anybody confirm that its ok with python 3.10.5?

Parvfect, 2022-08-03:
So it does not error out, but any operation after found muse device is not done and it ends running.

[image]

Andrey1994, 2022-08-03:
in your script there is only prepare_session

Parvfect, 2022-08-03:
> in your script there is only prepare_session

No for that specific screenshot for python 3.10.5 I'm running the signal quality check which normally checks the signal variance of each electrode and definitely gives an output after prepare session.

Parvfect, 2022-08-03:
> So it does not error out, but any operation after found muse device is not done and it ends running.
> 
> [image]

That one

Andrey1994, 2022-08-03:
could you try one of examples from brainflow docs?

Parvfect, 2022-08-05:
Tried following code sample adapted from the docs

```python
import argparse
import time

from brainflow.board_shim import BoardShim, BrainFlowInputParams, BoardIds, BrainFlowPresets

BoardShim.enable_dev_board_logger()
params = BrainFlowInputParams()
board = BoardShim(BoardIds.MUSE_2_BOARD, params)
board.prepare_session()
board.start_stream ()
time.sleep(10)
# data = board.get_current_board_data (256) # get latest 256 packages or less, doesnt remove them from internal buffer
data = board.get_board_data()  # get all data and remove it from internal buffer
board.stop_stream()
board.release_session()

print(data)
```

You're right it works as it should, so is functional with Python 3.10.5. Got the right output etc. 
[image]

But something really interesting happened after that. I tried installing eeg-notebooks which obviously failed because it's not compatible with Python 3.10.4, but after than even my brainflow stopped working with the same error 
[image]

So I tried creating a new conda env with python 3.10.4 and the same problem persisted for a while and then went away. There has to be something that is happening in the installation of eegnb that is interfering w brainflow on Windows 11 probably? Some common package that is causing issues? Cause it worked perfectly until I tried installing eegnb and then the same error persisted for a while and then it became okay again. This is all really confusing.

Andrey1994, 2022-08-05:
Unfortunately I am not an expert in eeg notebooks and dont know what it does internally and which packages it uses...

Andrey1994, 2022-08-09:
<< So I tried creating a new conda env with python 3.10.4 and the same problem persisted for a while and then went away.

it also can be that device stuck in a wierd state.. try to reboot it and see if there is an issue.

The only solution I see is to bump eeg notebooks to python 3.10 or use bled dongle

closed 2022-08-09 by Andrey1994